EVNotify/EVNotify #29: EVNotify for setups using wifi instead of Bluetooth

JGF2017, 2017-08-04:
Hi There. Great project, congratulations !

I wonder if it would be possible to expand the usage from older-versions of Bluetooth connectors into the more recent wifi-based ones... I am sorry I'm not capable of coding, otherwise I would try to help directly.
thanks
Joao

GPlay97, 2017-08-05:
Hey,

thank you.
Technically it would be possible - nevertheless it requires a complete new code base and handling for bluetooth communication. But the real problem is another point - and that is the reason why EVNotify is supporting bluetooth only:
If you use a Dongle via WiFi - the Dongle is the "hotspot" itself - so it is an "own" network - and your (mobile-) device which has EVNotify installed and has to communicate with the dongle, has to connect to this network. And from that moment on you can't reach the "normal internet" - you can't send or receive messages. And that would prevent from sending any requests to my server  - so you can't be notified.

So - to conclude - technically it would be possible - but it will not be supported because if you connect to a wifi dongle, the connected device can't use the normal internet - so with that it can't respond or send requests.

closed 2017-08-13 by GPlay97

ayasystems, 2019-04-19:
Hi

Please add wifi support.

I'm ussing a mikrotik to have an internet connection. My OBD2 can be use as client not only hotspot. 

I'm ussing Torque over wifi and work very fine!

Could be a great feature has a wifi option.

Regards!