EVNotify/EVNotify #34: When phone goes to sleep, communication stops

margarett, 2017-09-29:
Hi and thank you for your work in this applcation.

I am trying it with my new Ioniq, I am now able to see the current SOC, get email notifications and even use Telegram. Great :smile: 

The first problem I noticed is that, as soon as the phone shuts down the screen, the application does not send new requests to the OBD adapter (it has an activity LED which stops blinking at the set update rate when the phone goes to "sleep")

Of course I can configure the phone to not go to sleep, but that's a massive power waste.

Thanks

GPlay97, 2017-09-30:
Hello @margarett.

Thank you for using EVNotify and glad to hear that you've setup it successfully.

The phone screen can be turned off, this is a wanted feature from EVNotify - which works on my and several other phones without any problems. Requests are sent as if they would do when the screen is turned on.

So I think, that this is a specific problem with your phone. Which phone (and android Version) are you using? This sounds like a battery save mode from your phone which prevents EVNotify (process) to run in the background.

margarett, 2017-10-01:
I'm using a Xiaomi Mi4, Android 6.0.1
When I fix the other issue I'll check the phone settings in detail, then I'll report here

Bagger1896, 2018-02-11:
I have the same issue with Honor 6X

GPlay97, 2018-04-04:
Now there is a keepAwake Mode setting integrated within settings page in EVNotify. If your phone was killing the task previously, with keepAwake it should now work, since the device will be prevented to fall asleep (screen then always on).

Re-create an issue, if issue still persits.

Commit: d956ce55fa1998c03f2c577200a7929a6c0b3c16

closed 2018-04-04 by GPlay97