EVNotify/EVNotify #35: Communication with the car "dies" when the car is shut down

margarett, 2017-09-29:
Probably not an EVNotify issue, but I'd like to see some other feedback. When the car is shut down the OBD adapter seems to loose communication with the car, it remains powered but the application will not update the SOC anymore (and Telegram gets a notification stating the problem)

GPlay97, 2017-09-30:
Hello @margarett.

This is a normal behavior. The obd2 interface from the car will be powered-off when the car is shut down. The communication is then no longer possible.

To use EVNotify the car must be powered on - or - it must be charging. During charging, the interface of obd and the communication will be kept alive so EVNotify can fetch the state of charge.

The telegram notification was correctly sent out - because the communication was interrupted. During a normal charge, the connection is alive, as i said - if the charge interrupts - for whatever reason - EVNotify detects this charging error due to failed connection to the obd interface.

margarett, 2017-10-01:
Hi.
Thank you for your reply. Actually it is strange, today I was quick charging but I was unable to establish connection with EVNotify, although I was able to connect with Torque Pro without any issues (and of course I shut down Torque before attempting :grin: ). I enabled debug and got an error which might help
[image]

GPlay97, 2017-10-03:
Thank you for providing debug information.

The car responded with "CAN ERROR", so the interface of the car was not ready or could not handle the requests which were sent to the obd2 interface.

This is an error, which sometimes occurs (not often, but possible) - actually, to be honest, I'm not sure, why this can happen sometimes. It seems to be okay after a few minutes again - or after several restarts of the car or the app.

The CAN ERROR message is the response from the car to the state of charge request.

Hopefully, I'll have this message by myself, so I can inspect it by my own.

Does it work now again?

GPlay97, 2017-10-05:
Hello,

today I got the same error - it occurred after I've turned on and off the car while charging.
It was fixed and worked again when I re-established the connection (menu bar - Overview forces a reconnection).

I think the CAN ERROR comes directly from the car, when the interface stopped working for a short time (for whatever reason).

Maybe I can fetch this kind of error to force a reconnection automatically.

margarett, 2017-10-09:
Hello

Yes, I discovered that too some days ago. It seems that when you turn off the car while charging the communication between the ELM and the CAN bus dies. My OBD has some LEDs and it's clear that there is Bluetooth activity so the app is working, it's the CAN bus that goes down. Attached I send you screenshot of me turning off an on the car, the communication goes OK/ERROR the same way.
[image]

I guess that it must be something that the software should be able to handle because, as I told you, I'm using Torque Pro to access the BMS data (https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs) and the connection never "dies" in this program.

Thank you

GPlay97, 2017-10-10:
Thank you for your answer. I'm looking forward to add automatic error handling for this to re-establish the connection as soon as possible. I will create an issue within the next days and can hopefully fix this.
Thank you for the report and the information! This helps me a lot.

GPlay97, 2017-10-31:
@margarett CAN Error detection had been implemented in the newest update within commit 45cf87d776c4a0696c55875778823c4a535c8750. If CAN error detected, the interface will be re-established automatically :)

closed 2017-10-31 by GPlay97