EVNotify/EVNotify #38: How to use the Debug Mode to help you?

Gatam4U, 2017-10-11:
Hello,

many times I get a Telegram message :"Hallo! Es scheint so, als sei der Ladevorgang abgebrochen worden oder es gab einen Fehler bei der Kommunikation mit dem Auto. Am besten mal nachschauen!"

But the charging still work. How can I use the Debug Mode to send you Infos?

GPlay97, 2017-10-11:
Hello,

do you get this every time or only sometimes? 
After you are receiving this error in telegram - is the soc cycle in EVNotify still increasing? Or is it "frozen"? Anyhow - if you face this kind of error again - please click the debug icon in the overview page once. After a few seconds, there should be some data. Just screenshot it and upload it here.

Here is a little image of how you can enable the debug mode:
[Enable debug mode](https://evnotify.de/img/Debug.png)

What was the state of charge when this happened? 100%? Or earlier? The car was charging, wasn't it? Did you turn the car off while charging for a few seconds? This currently leads to some unexpected behavior, which will be fixed within an update.

I'm looking forward to help you :)

Gatam4U, 2017-12-23:
Sorry, that you don't get a answer from me.
I think this problem was with an older version. The last times this doesn't happen again.

GPlay97, 2017-12-23:
@Gatam4U No problem. Glad to know.
In an upcoming data, debug data will be sent automatically sent to me after turning on the debug mode or at least will be written to a log file, so you can easily share such things.

closed 2017-12-23 by GPlay97