EVNotify/EVNotify #47: EVNotify sometimes fails to connect to the bluetooth adapter

umrath, 2017-12-07:
For unknown reason EVNotify does not connect reliably to the bluetooth adapter and just "sits" there then.
This happens mostly:
- when another app has used the adapter (but has been stopped)
- when the adapter is plugged in but the app has been running already
- the car was switched off for a while

It usually helps to restart the app. It would be better if the app could do on its won whatever happens when the app restarts.

GPlay97, 2017-12-07:
Thank you for the report. This is a known bug, which has been forgotten, to be honest.

Currently the app only monitors the connection, once it's connected directly in the beginning. If it fails (for some reason), the mechanism doesn't start.

This was first reported here: #37.

I will try to fix this as soon as possible!

GPlay97, 2018-01-26:
This has been fixed within commit f758b2e59d4be42dc49d7c1b310233697f1ef284 and will be published as a new update very soon.

closed 2018-01-26 by GPlay97