EVNotify/EVNotify #48: Keep device from falling asleep

umrath, 2017-12-07:
The app let's the device switch off the display (and possible let's the device fall asleep).

An option to enable an "always on"-mode would be nice to have. This mode should keep the display switched on and prevent the device from falling asleep.

GPlay97, 2017-12-07:
Hello @umrath ,

thank you for the idea. That's sounds cool. I think, an optional setting for that would be great. What do you think? So the user can decide, whether or not "always on" should be enabled.
Furthermore I think this should be a local setting only. So it will not be synchronised within your account , so it allows you to maybe set the always on mode on your phone, but disables it on your tablet.
What's your opion?

GPlay97, 2018-04-04:
KeepAwake Mode option added within commit d956ce55fa1998c03f2c577200a7929a6c0b3c16

closed 2018-04-04 by GPlay97