EVNotify/EVNotify #52: Telegram notification usage

Schaussi, 2017-12-17:
Hi,

I would like to use a separate mobile phone for EVNotify in my car (which always stays in the car).
The notification feature for Telegram seems to be very useful, but I could not find the right place where to enter the token yet.
What I would like to achieve is to get a telegram message on my (main) phone, this phone is already registered for Telegram.
* Does EVNotify need Telegram installed locally on the (car) phone where it is running? 
* Or does it use some Telegram notification service, where the Telegram app is only necessary on the recipient's phone?
At the moment, I can only subscribe to the EVNotify channel but do not see a possibility how or where to write the "/subscribe" command with the token.

Thank you,
best regards

GPlay97, 2017-12-17:
Hi,

thank you for creating the issue.

I'm working on a little guide on the first start in an upcoming update, so things get more understandable.

To answer your questions:
- No.
- Yes. The phone in the car pushes a notification to the server. The server then send a notification to telegram, so you can then get the notification on all connected devices, where telegram is installed. So in your case the recipient's phone. You don't need a local telegram installation the "in-car-phone".

You need to contact the EVNotifyBot (https://telegram.me/evnotifybot). There you have to enter /subscribe and enter the token (please don't share it with others or here).

I have written a little guideline for the telegram integration here: https://github.com/GPlay97/EVNotify/wiki/Telegram-Integration

If you still have questions, feel free to ask.

Schaussi, 2017-12-18:
Hi,
perfect, I had confused the telegram EVNotify bot link with the EVNotify channel (where I could not send the subscribe command, of course).
Thank you!

closed 2017-12-18 by Schaussi