EVNotify/EVNotify #54: Read more information while loading

Gatam4U, 2017-12-20:
hello,

With Torque I read information from the cell, temperature and different.

It would be nice to get this information in EVNotify and can look back cause there are no base for all the different charger.

And when the car and battery get older there are some information about Maximal and Minimal Cell with Voltage and more.

[image]
[image]

GPlay97, 2017-12-20:
Hello,

thank you for creating this feature-request.

You are absolutely right - this would be awesome!

Currently, I'm working on supporting the Hyundai IONIQ Hybrid and IONIQ PlugIn Hybrid. They require a new command - and a new calculation.

After that, I will try my very best to fetch those information.

Keeping those data and archive them is also a very good idea. But first, I think, it will just read out those information. Later on, it will save them and analyzes them.

Gatam4U, 2017-12-20:
Thx.

All Step by Step.