EVNotify/EVNotify #56: OBDII adapter connection issues

Schaussi, 2017-12-21:
Hi,

today I tried to connect to a Hyundai Ioniq EV.
The OBD connection basically seems to work (it asks for the adapter's BT code, then it says it's connected), but finally ends up with an error message.
Unfortunately, I did not see any more detailed error texts.
My current OBD adapter looks like this one (not sure if it's 100% this model): https://www.amazon.de/Vgate-Mini-Elm327-Bluetooth-Scan-Fahrzeug-Diagnosetool/dp/B00NFP8PZ4
I have also ordered a Konnwei KW902 adapter (not delivered yet).
Could you give me a hint what could be the problem?
Is a special OBD adapter model needed for EVNotify?

Thank you,
best regards

GPlay97, 2017-12-21:
Hi,

can you try the following steps:

- pair the device over the android device settings directly within the bluetooth section
- start the IONIQ, as you would drive (you can stay in park), but the "engine" should be completely on to verify that everything is "awake"
- plug in the dongle
- enable bluetooth on the android device
- start EVNotify - and wait a few seconds (to speed up the process, choose a small polling interval within the settings)
- is everything working now? If yes, then it probably was an error on pairing the device - it should only be paired within the android settings. From now on, you can monitor the state of charge, even when the car is turned off AND charging. If not resolved, jump to the next step
- Please click the "bug icon" on the top - after that, the debug mode is turned on. You will then see some data at the bottom of the page. Can you take a screenshot of it and post it here? In the next update, it will be sent automatically or logged within a file, so it's easier to share that data.

Thanks in advance!

Most OBD2 adapters should work. However, I've listed some dongle [here](https://github.com/GPlay97/EVNotify/wiki/Compatibility), but the KW902 should work fine.

Schaussi, 2018-01-17:
The KW902 came today, works without any problems, thank you!

closed 2018-01-17 by Schaussi