EVNotify/EVNotify #61: NO DATA issue

GPlay97, 2018-01-31:
### New issue
- Short description: Some dongle only return no data, with Torque the soc can be retrieved
- Detailed description:
It seems, that some dongle require even more configuration commands on resetDongle function. Eventually, we also need to adjust the baud rate.

- Can you reproduce it? (_If so, how?_): Has been reported by a few users. Not reproducable with every dongle.

- [ X ] critical issue? (_issue, which will affect multiple accounts, 
  an issue which prevents that you can use the app in a normal way,
  bug which makes app unusable, security breach*_) (Yes/No):
  
 <details>
  <summary>additional information</summary>
  

  (_such as AKey [**NEVER EVER SHARE YOUR TOKEN!**], images_ or debug data from overview -> debug icon, your phone and android version)

</details>

\* **For very critical security breaches, please mail directly at info@evnotify.de**

joostandre, 2018-02-01:
Not all requests are supported by every vehicle. Yours may not have data to give, or was never designed to respond to this PID (or SID). If you are sure that the vehicle should be responding to this request, then check that the ELM IC’s timeout has not been set too short. Try issuing a AT ST FF command so the chip waits a full second before declaring there is no data available. Maybe your ECU just has higher priority things to attend to, and it is taking longer for it to answer you.

GPlay97, 2018-02-03:
Thank you for trying to help me :+1: 

I've read the same thing here: elmelectronics.com/wp-content/uploads/2016/07/ELM327DS.pdf

That's a very good manual.

Unfortunately I'm currently using all those commands.

Some user still get NO DATA, but when they use torque pro, they can read out the state of charge.

So it seems, that there is still missing something

joostandre, 2018-02-03:
Maybe you can contact the Torque developer?

joostandre, 2018-02-06:
Hi,

So you're using protocol number 6 - ISO 15765-4 CAN (11 bit ID, 500 kbaud)?
The setting I use in Torque is 0 - automatic and then it finds number 6 as
working protocol.

I am convinced that other adapters will work.
So although it is very nice of you that you want to send me one for
testing, it would be very nice to have my own adapter working ;-).

Thank you for all your efforts!

Kind regards,
Joost

GPlay97, 2018-02-09:
Hi, thank you for the information.

I'm using the automatic protocol search (ATSP0). But in the newest beta-version update I've added two more baudrates (230.4kbps and 500kbps).

can you test if it's now working? If still not, any idea, what to try next?

joostandre, 2018-02-12:
Hi mister evnotify,
Unfortunately, no success yet.
Dit you try (or have you already build it that way):

*"check that the ELM IC’s timeout has not been set too short. Try issuing a
AT ST FF command so the chip waits a full second before declaring there is
no data available. Maybe your ECU just has higher priority things to attend
to, and it is taking longer for it to answer you. "*

I think the adapter might be quite a slow one.
Kind regards,
Joost

joostandre, 2018-02-12:
I have a completly other question you might be willing to answer:
Next week friday (23 feb) I will be going to a Landal park in the Eiffel,
300 km drive, with my family.
I want to charge at a Lidl in Mönchengladbach.
As I understand it right, this is fast charging.
Will that always work with every cable (1, 2 or 3-fase?), so can I charge
to 80% in 25 minutes?
Or is the cable attached to the charger at such chargingpoints like the
Lidl's?
Kind regards,
Joost

GPlay97, 2018-02-23:
@joostandre Ouoh, sorry! I've forgotten to answer! Fast chargers always have their own cables. Then you should be able to charge up in less than 30 minutes. But Lidl mostly has only 20kW as far as I know.
Hope the road trip was good !

Regarding to the NO DATA problem: I've detected the issue. Will be fixed within the next beta patch.
The timeout was set to maximum, but adaptive timing was still on, so the timeout was not forced.

closed 2018-02-23 by GPlay97