EVNotify/EVNotify #77: Lost User - ID, no revert -> spam email

nixosgit, 2018-05-22:
- Short description: 
While loosing the **login user id**, the email service is not stoppable anymore.

- Detailed description:
I set up a user account. While checking all settings, I accidental clicked on changed user-id. As there is no "cancle", I entered "something" in order to close the window. That was not usefull anyway ... The system "stored" the "something" user id for ever. Now I can not remember my old user ID. And I also can not revert. 
To solve this bug: I created a new account, which works fine.
But the bug now, I still get emails from the old account. Even if I still know my password for the old account, there is no way to revert and stop the email service.

- Can you reproduce it?:
Have not tried, but I think will be possible the same way.

- [ ] critical issue? (Yes/No):
  Yes -> Spamming emails, with no possibility to stop.

GPlay97, 2018-05-22:
Thank you for the bug report.
I will check and fix this. The backend prevented the switch, since "something" didn't exist as an account. But the app changed it locally.
However, it is strange - are you receiving duplicate mails?
Nevertheless, please send me a mail to info@evnotify.de with the new AKey, you've created, so I can fix this here until the bugfix is released.

GPlay97, 2018-09-16:
Daily summaries have now been modified to prevent spam. Only on valid data and data not older than one day will be sent out as a notification. https://github.com/GPlay97/EVNotifyBackend/commit/7985c13cc8d396df8e076b516876276419db9265

closed 2018-09-16 by GPlay97