EVNotify/EVNotify #90: Question : Hyundai Kona support - Esim OBD device support

asetyde, 2018-10-10:
Hi,
I'm embedded engineer with Kona EV, I see your project but I want do a similar but more simple system to monitor charge of vehicle.
I want use this scheme : 
OBD2 -> EMBEDDED SIM DEVICE -> SERVER/APP
OBD2-SIM DEVICE -> SMS/SERVER SERVICE
OBD2 with interface to embedded device esim -> to app server or direct to sms or similar (integrate with EVNotify ? )

But at first I need to understand how read from obd and what , can you help me ? I have iOS but also android in my lab

asetyde, 2018-10-10:
we can start from here and i find a support esim device
http://arduinodev.com/hardware/obd-kit/

asetyde, 2018-10-10:
https://freematics.com/store/index.php?route=product/product&path=20&product_id=82

asetyde, 2018-10-10:
As you can see I must use AT command of this interface to test preliminary functions of OBD, but at first I must understand if I try to access to battery status .. and how ...

GPlay97, 2018-10-10:
Currently it's completely unknown how to be able to interact with the Kona EV. All comands for the IONIQ won't work for the Kona. Unfortunately.
It must be reversed engineered, before the command can be sent.

asetyde, 2018-10-10:
Oh !! Mmm about ioniq where do you find or how you find OBD commands

GPlay97, 2018-10-10:
Basically, what has been reversed engineered by others. Here is a good overview: https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/blob/master/Ioniq%20EV/extendedpids/Hyundai_Ioniq_EV_BMS_data.csv

asetyde, 2018-10-10:
https://github.com/JejuSoul/OBD-PIDs-for-HKMC-EVs/issues/22

closed 2018-10-10 by GPlay97