EVNotify/EVNotify #116: Suggestion: Client mode

gody01, 2018-12-09:
- Short description: 
Make EVNotify Client to backend.

- Detailed description:
Idea is, to have phone in the car sending data to backend.
Than on the other phone one could set EVNotify to client mode with same ID and receive data from backend and display it in the same way phone in the car display.

gody01, 2018-12-09:
Nice. I tried it before but it seems credential was not OK.

closed 2018-12-09 by gody01

GPlay97, 2018-12-09:
Just one hint:
There is currently one limitation:
On the receiver phone you also need to specifiy a car and a valid obd2 device.
This will change very soon.
But once set, it will work :)

GPlay97, 2018-12-09:
Furthermore:
There is a webinterface (currently made from a user) - from there you can also see your data - and don't need to install the app on the other phone, just open the website.

You can find the url on the thread: https://www.goingelectric.de/forum/allgemeine-themen/evnotify-2-fernueberwachung-fuer-dein-elektroauto-t36647.html

gody01, 2018-12-09:
Excellent :)
I only have to figure out, why BT OBDII conneciton is not veray stable. On Torque works without a problem on EVNotify there seems to be problem with connecting and even when connect after a while it disconnects.

GPlay97, 2018-12-11:
It seems to be an issue with some devices when torque was opened previously.
You can try one thing to ensure that the bluetooth handling is stable in EVNotify:
Restart phone, use EVNotify (don't use Torque before). If it's then stable, then it's a problem by Torque and your device. Unfortunately I don't have this issue and don't have an idea, why it's a problem on some devices after Torque usage.