EVNotify/EVNotify #117: Support for 2012 Prius Plug-in

rubin110, 2018-12-14:
### New issue
- Short description:
Support for 2012 Prius Plug-in

- Detailed description:
I own a 2012 Prius Plug-In. It would be rad if I could help to get support for the first generation of Prius Plug-Ins into this app. What can I do? I've got an OBD reader and happy to try things. Thanks!

GPlay97, 2018-12-16:
I will check and get back to you within the next few days with hopefully a test version

rubin110, 2018-12-19:
@GPlay97: Thanks for the response. I am in China for work (and ironically at a factory and next to a line where they're building car infotainment systems with OBD ports all over the place) and wont be back home till the beginning of January. So I wont be able to offer any testing time till I get back. Which is to say, please take your time. Thanks!

rubin110, 2019-01-02:
I am back home and ready to test. 

So I have this OBD I dongle...

https://www.amazon.com/dp/B01C6CZ0SQ/ref=cm_sw_r_cp_apa_i_WgilCbSK4A8AT

It emulates ELM 327. Is that close enough?

rubin110, 2019-01-07:
I've got a load of free time this week to do any testing you need. I've seen a bunch of updates wiz by through the Play Store but am unsure if they have anything related to the Prius Plug-In. Thanks.

GPlay97, 2019-01-09:
No, no updates for the Prius PlugIn. There are several versions of the Prius, each of it can interact different. I will try something, hopefully it works. Can you see the state of charge somewhere, to give me this value?

Please write a mail to info@evnotify.de - hopefully I can create a test version for you (will be closed internally with you only).

rubin110, 2019-01-21:
I found the SoC here...

https://priuschat.com/threads/2012-2015-prius-phv-custom-pids-for-torque-android-app.104779/

`015B`

However I'm unable to get that data from my OBD dongle when the ignition is off. I've left a comment in that thread asking if that's even possible.

GPlay97, 2019-02-04:
It's normal, that data can't be retrieved, if ignition is off. Only works, when car on or charging.

You can install current pre-release version from here: https://github.com/EVNotify/EVNotify/releases/tag/2.0.8-patch.2

Choose Opel Ampere E as a car. Turn car on. Turn debug mode on. Leave it on for a few minutes. Turn it off. Write me your Akey via mail (info@evnotify.de).

rubin110, 2019-02-08:
@GPlay97: Just sent you an email. I tried it a couple of times. The app is really susceptible to background killing on my Pixel 3 and Android Auto kept on getting in the way. A couple of the sessions was while attempting to charge, and then one 15 minute drive where the app may or may not got background killed.

Let me know if you want more info. Thanks!

rubin110, 2019-03-18:
Hey so after our back and forth over email, I think this might be a bust? Looking at all the Prius available data through Torque, the vehicle doesn't provide anything in the realm of SoC when the car is off and charging. 

The only thing I can think of now is hooking something up with GPIOs to the charging indicator light and having that notify me some how if the state of the light has changed.