EVNotify/EVNotify #125: OBD 2 Adapter schalted 'aus'

luckyheiko, 2019-01-22:
### New issue
- Short description: 
Adapter geht aus, aber nicht mehr an.

- Detailed description:
Hallo
habe mir folgenden Adapter für meinen Kona gekauft
https://www.amazon.de/Bluetooth-Diagnoseger%C3%A4t-TONWON-Scanner-Adapter/dp/B06ZZZSH48/ref=sr_1_13?ie=UTF8&qid=1548154556&sr=8-13&keywords=obd2+bluetooth+adapter+pro

ich musste nun feststellen, wenn ich die App schließe gehen die LEDs (vorallem die rote) aus und ich kann den Adapter nur wieder starten wenn ich den Adapter kurz rausziehe.

Liegt das an der App? das diese dem Adapter sagt 'geh aus'?!
habe mehrfach versucht den Debug Modus zu aktivieren, leider weiß ich nicht ob Daten (die richtigen) übermittelt wurden.

- Can you reproduce it? (_If so, how?_):
ja, immer beim beenden der App

- [ ] critical issue? (_issue, which will affect multiple accounts, 
  an issue which prevents that you can use the app in a normal way,
  bug which makes app unusable, security breach*_) (Yes/No):
  
 <details>
  <summary>additional information</summary>
  

  (_such as AKey [**NEVER EVER SHARE YOUR TOKEN!**], images_ or debug data from overview -> debug icon, your phone and android version)

</details>

\* **For very critical security breaches, please mail directly at info@evnotify.de**

luckyheiko, 2019-01-22:
Vergessen 
Huawai smart P mit Android 8 und neuste Version2 von EVNotify (2,0,7/02)

luckyheiko, 2019-01-23:
Nachtrag: 
ich habe nun EVNotifiy zwei Tage beobachtet. 
Ich denke das 'alle' Anzeigen schlüßig sind, bis auf die letzte Seite mit den Temperaturen.

Hier steht bei mir schon zwei Tage -2 -2 und 0
selbst wenn ich Fahre ist hier keine Änderung zu sehen

(Debuggmodus immer noch an)

luckyheiko, 2019-01-31:
Hallo 
und vielen Dank für die Info. 
Finde es halt nur etwas merkwürdig, das der Dongel 'sofort' ausgeht.
Könnte ich irgenwie helfen dies zu testen? ohne den Dongel durch die gegend zu schicken?

(ist aber nicht wichtig, da ich nun den 'kleinen blauben' bestellt habe)

GPlay97, 2019-02-01:
Das ist gewollt von mir. Beim Beenden (über die "Back" Taste von Android) wird der Standby Modus forciert. Es obliegt allerdings dem OBD2-Dongle Hersteller, wie dieser implementiert ist, weshalb ich da nichts machen kann, wenn der Hersteller ihn vollständig abschaltet

closed 2019-02-18 by GPlay97