EVNotify/EVNotify #152: Charging data Kona EV not correct

matthiaslink77, 2019-05-05:
### New issue
- Charging data Kona EV not correct
- Detailed description:
Hyundai Kona EV charging screen (page 3) shows:
Battery current: -3.10A
Charge-/Discharge power: -1,26kW

Car is charging with 8A and approx. 1.7 kW

Estimated range also does not show the actual vehicle information.
App shows
[image]
[image]

 350/365 km, while car shows already > 400 km range at 96 % charge.

- Can you reproduce it? (_If so, how?_):
Yes. Just connect app and read data.

- [ ] critical issue? (_issue, which will affect multiple accounts, 
  an issue which prevents that you can use the app in a normal way,
  bug which makes app unusable, security breach*_) (Yes/No): No
  
 <details>
  <summary>additional information</summary>
  

  (_such as AKey [**NEVER EVER SHARE YOUR TOKEN!**], images_ or debug data from overview -> debug icon, your phone and android version)

</details>

\* **For very critical security breaches, please mail directly at info@evnotify.de**

GPlay97, 2019-05-05:
Hey,

thank you for contributing your first issue and the detailed information.

The charging data seems to be wrong in this case. On slack, another user faced a similar issue while charging on AC type2.

To inspect this, can you enable "debug mode" (press the little bug icon on the top) and let it enabled for a few seconds, while it fetches data from the car via Bluetooth. Enable it, when this happens again (should be reproducable).

This could help a lot. Then send me your data - either here - or via mail info(at)evnotify.de.

The range is just estimated based on the consumption you entered, in your case 17.5kWh/100km. Just simple maths currently, no "real data" or intelligent algorithms yet.

GPlay97, 2019-06-07:
After investigating, it comes out, that this is actually the battery data from OBD.

In your example -> 404,9V * 3,1A = 1255,19 W => 1,25519kW.

So actually this is not a bug, it's a feature ;)

Thanks also to @EVSalomon for finding out :)

closed 2019-06-07 by GPlay97