EVNotify/EVNotify #153: can't select a new bluetooth device on iOS 12.2 (iPhone & iPad)

rjwelling, 2019-05-09:
### New issue
- Short description: installed EvNotify from Appstore on both iOS 12.2 iPhone 8p and Ipad 9
- Detailed description: recommended Vgate ICar Pro can't be selected in the Setting, OBDII Device menus.

- Can you reproduce it? (_If so, how?_): Is reproducible at 100% rate 

- [ ] critical issue? (_issue, which will affect multiple accounts, 
  an issue which prevents that you can use the app in a normal way,
  bug which makes app unusable, security breach*_) (Yes/No):
  
 <details>
  <summary>additional information</summary>

As such, without BT device you can't read the codes. 

  

  (_such as AKey [**NEVER EVER SHARE YOUR TOKEN!**], images_ or debug data from overview -> debug icon, your phone and android version)

</details>

\* **For very critical security breaches, please mail directly at info@evnotify.de**

GPlay97, 2019-05-09:
Hi,

As mentioned in the compatibility list ( https://github.com/EVNotify/EVNotify/wiki/Compatibility ) directly at the beginning, the Bluetooth communication is still in early development for iOS.

"EVNotify is now also available for iOS, but currently, Bluetooth communication is not working yet. It's in development for iOS devices.".

Hopefully it will fully work within the next patch/patches. 

The listed dongle will be the first one that is fully supported.

rjwelling, 2019-05-09:
okie do, let me know if I can assist with testing BT on iOS.

/rw

McTrex, 2019-06-11:
As I only have a spare iOS phone I would also very much appreciate bluetooth working on iOS.

PerfectAnswer, 2019-07-04:
Hi,
same Story here with IOS and IPhone 7+.
Also use a Vgate ICar Pro. In the dongle documetation is a hint, related to the undetectability of the dongle. The dongle will not be showed in the bluetooth list but wenn I start a search with another app (AutoDoctor IOS), the the dongle will be detected.
The dropdown box in EV Notify contains not one single bluetooth device like on Android, so for me looks like there is a general problem with the bluetooth devices.
Hope it helps :)
BR
Bernd

GPlay97, 2019-07-13:
Bluetooth is now working on iOS. The next update for iOS will have it (after Apple reviewed and acccepted the changes)

PerfectAnswer, 2019-07-23:
> Bluetooth is now working on iOS. The next update for iOS will have it (after Apple reviewed and acccepted the changes)

Sounds great!
Uploaded to the store and checked by apple?

GPlay97, 2019-07-23:
It's currently under review. There were some things, that were rejected by Apple, that I need to modify. Should be available within next days hopefully.

PerfectAnswer, 2019-07-23:
Sounds great.
Just for the case, but if you need tester with IOS Devices, I am available.
Did you work with [TestFlight](https://developer.apple.com/testflight/)?  
It's easy to add tester before you send to the store review.

Pazekal90, 2019-07-23:
I’m  also available for iOS Beta testing via TestFlight

GPlay97, 2019-07-24:
Not working with TestFlight. It also requires a review, especially for the first time. Might not be that hard to pass, but same effort for me as a developer.
Especially when a lot of things changed, like now, there are so many things, that will be rejected. Pixel missing here, Metadata not set there. 
Long process. Every time.

PerfectAnswer, 2019-07-30:
Yeah! Apple loves all the small things like pixelperfect Icons for every device and resolution, metadata, country stuff etc. We are with you and wait still with patience :)

GPlay97, 2019-08-17:
The update for iOS is now available with working bluetooth :)

closed 2019-08-17 by GPlay97

PerfectAnswer, 2019-08-20:
Connection works now. Thanks.
But, if you stop the engine the connection is lost and after restart the car, the connection is still lost.
I've enabled the debugging mode, because the App shut down after a few minutes in use. Hope it helps a bit.