EVNotify/EVNotify #154: Can't create new account in apl

maulwuff, 2019-06-15:
### New issue
- Short description: 
App responds with `unexpected error` when hitting next on the password dialog. 

- Detailed description:
Enter password twice in the registration process and hit next. `Create account` label will turn red and show `unexpected error` beneath it. 

- Can you reproduce it? (_If so, how?_):
Try any password, as long as they match. 

- [x] critical issue?  (Yes/No): yes, prevents creating new accounts. 
  
Happens with Version from 03.2019 and betra from 05.2019

Android 8.1.0

GPlay97, 2019-06-15:
Hey,

this is currently a problem with the server. I'm working on it. Sorry for the circumstances. You can have a look at status.evnotify.de for more information.

maulwuff, 2019-06-15:
Oh, ok. Then I'll try tomorrow. (Thought I could have a nice evening with my e-niro in the garage, playing around with evnotify 🤣)

GPlay97, 2019-06-15:
I'm very sorry for this!

GPlay97, 2019-06-15:
All services are up and running again. Sorry for the trouble!

maulwuff, 2019-06-16:
Hm, still doesn't work, with the same error as yesterday. Status page says, everything is green.

GPlay97, 2019-06-16:
Since it was a DNS issue, it can take some time, until it's updated to everyone. Can you completely close the app, restart your phone and then try again? I tried to register new accounts, it works fine here.

maulwuff, 2019-06-16:
Is there a way to get a more detailed error message? I rebooted, tried wifi and cellular, uninstalled from beta to main release, tried in airplane mode to see what happens, but it always fails.

GPlay97, 2019-06-16:
Hm, normally an unexpected error message comes, when it can't be handled.

For further inspection, can you send me a mail to info@evnotify.de together with your AKey? Maybe I can find something out.

maulwuff, 2019-06-16:
Where can I find an Akey when I've not registered, yet? It fails on the second screen of creating an account.

GPlay97, 2019-06-16:
You are right.

Then just mail with exact details what you enter and where. 
Maybe attach also screenshots in mail.
Thanks.

maulwuff, 2019-06-16:
:-) just edited my last comment with  screenshot.

GPlay97, 2019-06-16:
No,because I can't reproduce it.

Are you in WiFi? Maybe try with mobile data. Furthermore - just for testing - try a password 123456 or so (at least 6 digits).
But please send a mail, this is better to get more information

maulwuff, 2019-06-16:
Make sure to not block the app in your firewall....

closed 2019-06-16 by maulwuff