EVNotify/EVNotify #156: ABRP integration not working, cannot reset the link

jwveldhuis, 2019-07-10:
### New issue
- Short description: no data shown in ABRP, reset token, unclear if I need to reset the ABRP integration and how to do that
- Detailed description:
I have no idea whether this is a ABRP issue or a EVNotify issue.
In web.evnotify.com in Setting, Integrations I clicked on the "A BETTER ROUTE PLANNER" button and granted access. Now I cannot reproduce the same and seeing the following message no matter what I try:
"You have linked your account with A better route planner."
I even tried to invoke the API with a `,"ABRP":null,` but the ABRP id string remains unchanged.
Is there a way to revoke the access to start over? Is there a way to confirm a correct integration between EVNotify and ABRP?

GPlay97, 2019-07-10:
For ABRP integration you will need location synchronization enabled to submit the location data to ABRP.

You will be able to remove integration soon.

jwveldhuis, 2019-07-10:
Ah thanks, will try that! So resetting the token is not affecting the ABRP integration, even though it mentions 
>(...) Resetting your token disconnects every app and service. (...)
When I click on Reset Token?

jwveldhuis, 2019-07-10:
Anyway, I just tested with the Location sync enabled and it is working now!
Thank you for creating this app, excited to use it on a first long road trip!

Pazekal90, 2019-07-10:
How is location sync activated for ABRP?

jwveldhuis, 2019-07-10:
@Pazekal90 I enabled 'Location Synchronization' in the Settings of the EVNotify Android app. Had already linked to ABRP. Then the data started to show in ABRP after I connected the app with the car.

Pazekal90, 2019-07-10:
And how can I do this on iOS without the app? I have only installed the PWA web.evnotify.de

GPlay97, 2019-07-13:
Currently, to submit data, you have to use the Android App to send location data. ABRP requires location data.

Pazekal90, 2019-07-13:
@GPlay97: Ah ok is there a timeline to add this feature to iOS. I think this would be one of the main features iOS users will use to get instant route planning with live data from car.

GPlay97, 2019-08-13:
Resetting token via ABRP API is now in backlog. This is currently not implemented in their API. Once implemented, it will be integrated.

joncox123, 2019-09-10:
This also isn't working for me with the latest EVNotify on Android as installed via Google Play Store. I'm going to try to dig into it, but some explanation of how this is supposed to work would be appreciated.

GPlay97, 2020-12-11:
Bolt and Ampera were integrated in commit https://github.com/EVNotify/EVNotifyBackend/commit/e8febce169ed3ef1724348d25feb99e6f4f1e913

Resetting will be tracked in #204

closed 2020-12-11 by GPlay97