EVNotify/EVNotify #159: Building EVNotify with cordova / gradle not working

joncox123, 2019-09-11:
I am not able to build EVNotify with cordova. I have to admit, I'm more of a make, cmake, Android Studio, Java and C/C++ kind of guy...but I'd like to add full support for the Chevy Bolt / Opel Ampera-e.

I followed the instructions in CONTRIBUTING, plus I downloaded the google-services.json file by creating a Firebase project with Google (this seems to be missing from the build instructions, since cordova wouldn't proceed without it).

From what I gather, this may have something to do with not building with the correct android sdk version. But which build.gradle file am I supposed to set this in? There are like 5 different seemingly superfluous versions floating around in the build directory.

For reference, I'm using Ubuntu 19.04 with gradle 4.4.1, OpenJDK 8 and cordova 9.0.0.

**When I run 'cordova build android', it fails with:**

> joncox@desktop:~/git/EVNotify/app$ cordova build android
> ANDROID_HOME=/home/joncox/Android/Sdk
> JAVA_HOME=/usr/lib/jvm/java-8-openjdk-amd64
> Subproject Path: CordovaLib
> google-services plugin could not detect any version for com.google.android.gms or com.google.firebase, default version: 9.0.0 will be used.
> please apply google-services plugin at the bottom of the build file.
> Configuration 'compile' in project ':' is deprecated. Use 'implementation' instead.
> The Task.leftShift(Closure) method has been deprecated and is scheduled to be removed in Gradle 5.0. Please use Task.doLast(Action) instead.
> 	at build_bxjyaiqcdwawufwq8ds61o00x.run(/home/joncox/git/EVNotify/app/platforms/android/build.gradle:144)
> registerResGeneratingTask is deprecated, use registerGeneratedFolders(FileCollection)
> registerResGeneratingTask is deprecated, use registerGeneratedFolders(FileCollection)
> publishNonDefault is deprecated and has no effect anymore. All variants are now published.
> :preBuild UP-TO-DATE
> :CordovaLib:preBuild UP-TO-DATE
> :CordovaLib:preDebugBuild UP-TO-DATE
> :CordovaLib:checkDebugManifest UP-TO-DATE
> :CordovaLib:processDebugManifest UP-TO-DATE
> :preDebugBuild UP-TO-DATE
> :CordovaLib:compileDebugAidl UP-TO-DATE
> :compileDebugAidl UP-TO-DATE
> :CordovaLib:packageDebugRenderscript NO-SOURCE
> :compileDebugRenderscript UP-TO-DATE
> :checkDebugManifest UP-TO-DATE
> :generateDebugBuildConfig UP-TO-DATE
> :prepareLintJar UP-TO-DATE
> :generateDebugResValues UP-TO-DATE
> :generateDebugResources UP-TO-DATE
> :processDebugGoogleServices
> Parsing json file: /home/joncox/git/EVNotify/app/platforms/android/google-services.json
> :CordovaLib:compileDebugRenderscript UP-TO-DATE
> :CordovaLib:generateDebugResValues UP-TO-DATE
> :CordovaLib:generateDebugResources UP-TO-DATE
> :CordovaLib:packageDebugResources UP-TO-DATE
> :mergeDebugResources UP-TO-DATE
> :createDebugCompatibleScreenManifests UP-TO-DATE
> :processDebugManifest UP-TO-DATE
> :splitsDiscoveryTaskDebug UP-TO-DATE
> :CordovaLib:platformAttrExtractor UP-TO-DATE
> :CordovaLib:processDebugResources UP-TO-DATE
> :processDebugResources/home/joncox/.gradle/caches/transforms-1/files-1.1/core-1.0.0.aar/796462d369df4cd4cb5b7d02d5a58016/res/values/values.xml:133:5-70: AAPT: error: resource android:attr/fontVariationSettings not found.
>     
> /home/joncox/.gradle/caches/transforms-1/files-1.1/core-1.0.0.aar/796462d369df4cd4cb5b7d02d5a58016/res/values/values.xml:133:5-70: AAPT: error: resource android:attr/ttcIndex not found.
>     
> /home/joncox/git/EVNotify/app/platforms/android/build/intermediates/incremental/mergeDebugResources/merged.dir/values/values.xml:95: error: resource android:attr/fontVariationSettings not found.
> /home/joncox/git/EVNotify/app/platforms/android/build/intermediates/incremental/mergeDebugResources/merged.dir/values/values.xml:95: error: resource android:attr/ttcIndex not found.
> error: failed linking references.
> 
> Failed to execute aapt
> com.android.ide.common.process.ProcessException: Failed to execute aapt
> 	at com.android.builder.core.AndroidBuilder.processResources(AndroidBuilder.java:796)
> 	at com.android.build.gradle.tasks.ProcessAndroidResources.invokeAaptForSplit(ProcessAndroidResources.java:551)
> 	at com.android.build.gradle.tasks.ProcessAndroidResources.doFullTaskAction(ProcessAndroidResources.java:285)
> 	at com.android.build.gradle.internal.tasks.IncrementalTask.taskAction(IncrementalTask.java:109)
> 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
> 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
> 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
> 	at java.lang.reflect.Method.invoke(Method.java:498)
> 	at org.gradle.internal.reflect.JavaMethod.invoke(JavaMethod.java:73)
> 	at org.gradle.api.internal.project.taskfactory.DefaultTaskClassInfoStore$IncrementalTaskAction.doExecute(DefaultTaskClassInfoStore.java:173)
> 	at org.gradle.api.internal.project.taskfactory.DefaultTaskClassInfoStore$StandardTaskAction.execute(DefaultTaskClassInfoStore.java:134)
[... 288 lines cut ...]
> 	at com.google.common.util.concurrent.AbstractFuture.get(AbstractFuture.java:462)
> 	at com.google.common.util.concurrent.AbstractFuture$TrustedFuture.get(AbstractFuture.java:79)
> 	at com.android.builder.internal.aapt.v2.QueueableAapt2.lambda$makeValidatedPackage$1(QueueableAapt2.java:179)
> 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
> 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
> 	... 1 more
> Caused by: com.android.tools.aapt2.Aapt2Exception: AAPT2 error: check logs for details
> 	at com.android.builder.png.AaptProcess$NotifierProcessOutput.handleOutput(AaptProcess.java:454)
> 	at com.android.builder.png.AaptProcess$NotifierProcessOutput.err(AaptProcess.java:411)
> 	at com.android.builder.png.AaptProcess$ProcessOutputFacade.err(AaptProcess.java:332)
> 	at com.android.utils.GrabProcessOutput$1.run(GrabProcessOutput.java:104)
> 
> FAILURE: Build failed with an exception.
> 
> * What went wrong:
> Execution failed for task ':processDebugResources'.
> > Failed to execute aapt
> 
> * Try:
> Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
> 
> * Get more help at https://help.gradle.org
> 
> BUILD FAILED in 0s
>

GPlay97, 2020-01-02:
Welcome to the hell of Cordova and Gradle ;)

Unfortunately this really depends when you first created your builds and which versions are used. 
You need an older version, I think something like Cordova 6.
It will not work with latest versions.

Please try downgrading also the android platform version within cordova.

It is always a try and error unfortunately. Because it always varies depending on which gradle, sdk, android version the plugin developers uses (and changes).

gertst, 2020-02-08:
Same issue here ... I'll try with some older versions and will report back the working combinations ;-)

ektus, 2020-03-23:
I don't have (and won't get) a google developer account. Without that, I seem to be stuck as google-services.json is missing and cordova refuses to do anything without it. Is there a way around this?
It should be possible to build open source applications without google services, e.g. to publish them as apk or on f-droid.

bje-, 2020-06-02:
Thanks @ektus. I got a bit further using Cordova 6.0 until:

```Error: cordova-plugin-fcm: You have installed platform android but file 'google-services.json' was not found in your Cordova project root folder.```

GPlay97, 2020-06-02:
The latest Cordova is not working. I had struggle a long time to build it. Shortly I made it again. I could post the configuration later.

For the Google FCM Plugin (which is used for notifications) the service json file is required (only applicable if you have a google account and registered there).

You can still build it, if you just remove the fcm plugin from the plugin reference file, then it will no longer be used.

closed 2020-12-11 by GPlay97

b-reich, 2021-08-31:
@GPlay97  which Version is working?