EVNotify/EVNotify #165: e-Niro misses CEC and CED

maulwuff, 2019-11-02:
I have a e-Niro and when I choose this as vehicle in the app, CEC and CED are missing (shows 0).
When I use Kona, all fields are filled. 
Something seems to be wrong in the e-Niro profile, which should be the same as in Kona ;-)

GPlay97, 2020-01-02:
Will be available soon

ektus, 2020-03-20:
How soon is "soon", especially on Android?

GPlay97, 2020-03-21:
https://github.com/EVNotify/EVNotify/issues/166#issuecomment-599259624

GPlay97, 2020-05-12:
@ektus @maulwuff 
Sorry for the huge delay. App build works again..

It was implemented. If you still encounter issues, let me know in a new issue.

closed 2020-05-12 by GPlay97

ektus, 2020-05-13:
I've got the update, and it looks good now with my 2019 e-Niro. No more need to spoof Kona :-)

commit e2cede31 (2020-05-12): add cec, ced and bugfixes for e-Niro
- closes #165