EVNotify/EVNotify #166: Hyundai Ioniq 38kw 2019

MartijnSanders14, 2019-11-10:
Can't connect with new Hyundai Ioniq 2019 Electric
- App gives 'invalid date' when I try to connect with the car. App is giving the range of the old Ioniq model. 

I use an Iphone 11 Pro Max with latest updates and a Vgate iCar Pro Bluetooth 4 OBD2. 

[image]

noradtux, 2019-11-10:
The 38kWh Ioniq is not supported yet, we don't have access to one for testing. You might want to try to set Kona as car. There is a good chance the new Ioniq behaves the same.

MartijnSanders14, 2019-11-10:
Good to know. Gonna check it!

MartijnSanders14, 2019-11-12:
Okay, good news the Ioniq is working with the Kona. Only the battery capacity is different. 
Is that easy to change/setup?

[image]
[image]
[image]

GPlay97, 2019-11-12:
Cool, thanks for the information.

Yes, capacity can be changed very soon within an update

MartijnSanders14, 2019-11-12:
If you need someone to test, just send me a message!

MartijnSanders14, 2019-11-13:
For the meantime I changed the consumption to 24,08 kWh/100 so the range is correct.

MartijnSanders14, 2019-12-13:
Hi, any idea when you will come up with an update? I'm looking forward to it.

GPlay97, 2019-12-13:
Hi,

yes, this weekend.

I had a surgery and was not able to work. I'm back now ;)

MartijnSanders14, 2019-12-13:
Ah thanks for your reply. Everything went well?

I’ve got an other question, with an android phone, EVnotify keeps the bluetooth 4.0 adapter online and active. My iPhone 11 is stopping after 30 mins with the connection (same connector). Anything that you can do about that or is that in to the source code of Apple?

Kind regards, 

Martijn

SiGmAX666, 2019-12-29:
I have the same issue as @MartijnSanders14 regarding disconnect after some amount of time, is there a fix for that? Is it possibly the FW of the vgate adapter?

MartijnSanders14, 2019-12-29:
I solved it with an android phone. Bought a cheap phone to connect the car with EVNotify. The android phone sends the data to A better routeplanner. At my iPhone I use the routeplanner.

But all together there are two problems for now:
1. Unstable Bluetooth connection with EVNotify and some adapters. Known problem, has to do with keeping alive the app and thereby the connection. Especially in the last IOS is keeping alive of apps a problem. It is fixable. 
2. Evnotify is not updated and functioning for the Ioniq 38kWh. I used the Kona 64kWh, but the charging dynamics etc used are different. So ABRP is not working well. So I hope there will be an update really really soon.

GPlay97, 2020-01-02:
Yes, update soon. Due to christmas and new year holidays, it delayed more than I expected ^^

On iOS it freezes, this is known. Apple restrictions currently (you need to get permissions from them to run in background, this is currently forbidden for me by them...). iOS update is unlikely to be available soon.

For android this is nearly always energy saving options from phone. Either find them and turn them off, or go to  EVNotify settings, down to version number, click it multiple times very fast, enable then background mode + persistent notifications, save. Then it does everything for you automatically and runs in background, as long as EVNotify app is opened (even with screen locked).

MartijnSanders14, 2020-01-16:
Thanks for your replay. What is your planning for the update? I’m looking forward to it.

nicvhs, 2020-01-19:
I have 1 question regarding the Kona Ev 64kWh. I was testing with an android with 4G in the car with Evnotify running fine and then i connect through web or other phone with app to read out the data at a distance. It looks like the ODB2 dongle goes toe sleep when the car turns off.
I still see a light blinking on the ODB2 but Evnotify loses connection. Any wyas to solve this that you know of? I tried it also while attached to a public charger.

thanks
Nic

noradtux, 2020-01-19:
The Kona (and most other cars I know of) stops responding to commands when turned off as the control units shut down. So there is no data to transmit.

MartijnSanders14, 2020-02-06:
Any progress on the update? I’m really looking forward!

nicvhs, 2020-02-07:
No:(

The kona powers off the odb plug...
So, it worked when the car was turned on but when turned off it doesn’t work:(

I am now looking for a system to put on the charging cable. We should be able to measure the current flow and get a start/stop of current

Kind regards
Nicholas van Hoey Smith

MartijnSanders14, 2020-02-07:
Sorry, I meant @noradtux for answering about the update for the Ioniq 38kw. It’s still only the 28 and that script is not functioning. I’m driving the car since 09-19 and the missing of EVNotify is really annoying.

GPlay97, 2020-02-07:
I'm the responsible person for this update, not @noradtux .
Since I had a lot of stuff to do due to the new PlugAndPlay prototype, I couldn't publish the update. But since yesterday, the prototype seems stable, so it looks like I can focus on the app this weekend finally again!

MartijnSanders14, 2020-02-07:
Ah sorry that I confused. Thanks for the status update.

eChrisPa, 2020-02-13:
I am interested in the IONIQ FaceLift too.

Additionally I am using ABRP and I could help to verify the abrp integration with new IONIQ profile. I could try beta apk if necessary ...

MartijnSanders14, 2020-02-18:
We can both test, just let us know when we can/have to start testing. Half March I’m going to do a 1500km trip, would be great if it’s functioning at that moment.

JoaquimLopes1980, 2020-03-01:
> I'm the responsible person for this update, not @noradtux .
> Since I had a lot of stuff to do due to the new PlugAndPlay prototype, I couldn't publish the update. But since yesterday, the prototype seems stable, so it looks like I can focus on the app this weekend finally again!

Hello. If you need a tester i can help. Have one ioniq 38Kwh. If you need just say.

MartijnSanders14, 2020-03-10:
> I'm the responsible person for this update, not @noradtux .
> Since I had a lot of stuff to do due to the new PlugAndPlay prototype, I couldn't publish the update. But since yesterday, the prototype seems stable, so it looks like I can focus on the app this weekend finally again!

Hi, anything about the update? Next week I'm in kind of need of the function of your wonderful app. Do you think that it's possible to use it with the Ioniq 38kw?

GPlay97, 2020-03-15:
My android build is unfortunately completely destroyed.. They changed something again and my build is no longer working, neither locally nor via Cloud container.

I have no clue, when I can fix this weird thing.

ektus, 2020-03-23:
I do have a working build environment for Android to build LineageOS, but I'm not a programmer (LOS build is fully automatic). Are there any build instructions for EVNotify, and is the code for Android current?

GPlay97, 2020-04-28:
No, I tried again this weeks. It looks like the images / CDNs I used doesn' exist any longer, but I need them to compile properly. I'm currently working on new native app.

GPlay97, 2020-05-12:
Finally.

4ef501d2f921c8e3186cde5861c1ca1cd1c59ce4

Implemented. Live soon in playstore

closed 2020-05-12 by GPlay97

eChrisPa, 2020-05-13:
Thanks for implementing the IONIQ FL, quite a lot of code for the new car.

Unfortunately, it does not work yet. When I select and store the new car in my publishing mobile or in my subscribing mobile, the app says "NO_CAR_SELECTED". 

I had to revert back to KONA with 39kWh bat.

GPlay97, 2020-05-13:
Thanks for the feedback. I'll look at this later, maybe I missed something

JoaquimLopes1980, 2020-05-14:
> Finally.
> 
> [4ef501d](https://github.com/EVNotify/EVNotify/commit/4ef501d2f921c8e3186cde5861c1ca1cd1c59ce4)
> 
> Implemented. Live soon in playstore

Hello. Where can i find it? 
Thanks for your work.

ektus, 2020-05-15:
https://play.google.com/apps/testing/com.evnotify.app

GPlay97, 2020-05-17:
@eChrisPa I fixed it. Uploaded it. It will be available shortly . I will also update the apk later under releases here

MartijnSanders14, 2020-05-17:
Any forecast for the App Store?

eChrisPa, 2020-05-17:
@GPlay97 

Wow, this was a fast fix. Thanks!

GPlay97, 2020-05-17:
@MartijnSanders14 Not yet, a bit more complicated process (and apple needs multiple days until they release a patch). But it's planned, yes.

@eChrisPa Was only a small thing, could be even faster, but just had now the time for it. So it works fine now? :)

eChrisPa, 2020-05-17:
I saw the git commit, but don't have a release.

I'm a (windows) developer, would it be complicated for me to build the Android APK myself?

Or is there some automatted build pipeline on github which will throw out the APK?

eChrisPa, 2020-05-17:
Got the download from PlayStore this second.

Installed it on sender andf receiver mobile.

While the max. range looks plausiblle now (444km -> 263km ) all other values are 0.

Tried to start/initialize several times. The values stay zero. 

One note: When selecting the new "IONIQ FaceLift EV" car, a red warning shows up "Daten werden ggf. falsch interpretiert" which means  "data might be misinterpreted" or so ...

GPlay97, 2020-05-17:
@eChrisPa Please then press the debug icon on the phone that is connected to the dongle within dashboard. Let it on for half a minute or so. Click again to deactivate it.

Write me akey (per mail info@evnotify.de)

lepinkainen, 2020-09-18:
I have the same issue with my MY20 Ioniq Electric. Everything except for max range works just perfectly with the Kona EV setting.

Any chance of getting this fixed? Maybe just copy the whole Kona profile over the Ioniq one to make this work even somewhat.

MartijnSanders14, 2020-09-18:
Use an android phone, become a beta tester (you can select that in the playstore). You can select the Ioniq 38. I asked ABRP to enable EVNotivy for the Ioniq 38. Works perfectly, used it in the last 6 weeks for 5400km. Had no problems.