EVNotify/EVNotify #172: Kia e-Soul support missing

gidhap, 2020-01-08:
### New issue
- Short description:  Kia e-Soul support missing
- Detailed description: I just downloaded the app, but unfortunately my Kia e-soul is not listed (yet?)
Also I'd like to donate, but not on a monthly base, th eoption for a one off paypment via PPal would be good to have.
- Can you reproduce it? (_If so, how?_): 

- [ ] critical issue? (_issue, which will affect multiple accounts, 
  an issue which prevents that you can use the app in a normal way,
  bug which makes app unusable, security breach*_) (Yes/No):
  
 <details>
  <summary>additional information</summary>
  

  (_such as AKey [**NEVER EVER SHARE YOUR TOKEN!**], images_ or debug data from overview -> debug icon, your phone and android version)

</details>

\* **For very critical security breaches, please mail directly at info@evnotify.de**

noradtux, 2020-01-09:
You could try to select the Kona in EVNotify. So far Kona, e-Niro and Ioniq FL behave the same. If Kona works for you adding support for the e-Soul should be trivial.

gidhap, 2020-01-09:
Txs, I'll try as soon as my OBD adapter has arrived

GPlay97, 2020-01-09:
@noradtux @gidhap 
It was confirmed, that it works with the Kona recently!

It will be officially added in the next update!

GPlay97, 2020-05-12:
Added.

Will be live in the update published later.

commit 5a910a31 (2020-05-12): add full support for e-Soul
- closes #172
closed 2020-05-12 by GPlay97