EVNotify/EVNotify #173: Daten Export direkt in ABRP?!

luckyheiko, 2020-01-09:
Hallo 

ich habe heute mal versucht den ABRP und EVNotify (beides als APP) zum laufen zu bekommen. 
wenn ich das nun richtig verstanden habe, tut der ABRP nur Daten aus dem 'Netz' nehmen (Logs)
(bei Torque und bei EV Notify)

mein erstes Problem ist hier das ich zur Zeit den ABRP und das EVNotify nur in der Web Version nutzen kann (muss) damit ich diese Live Daten nutzen kann.

zweites Problem, wo Daten rauf und runtergeladen werden entsteht Traffic und ich z.B. habe einen Tarif der keine unbegrenzte Datenmenge zur Verfügung hat.

Ist es angedachte mit ABRP und EVNotivy eine 'direkte' Verbindung herzustellen?
Also APP zu APP und nicht über den Webbrowser

oder ist dies schon möglich und ich habe es noch nicht gesehen?

würde mich über eine kurze Info sehr freuen.

mfg heiko

### New issue
- Short description: 
- Detailed description:

- Can you reproduce it? (_If so, how?_):

- [ ] critical issue? (_issue, which will affect multiple accounts, 
  an issue which prevents that you can use the app in a normal way,
  bug which makes app unusable, security breach*_) (Yes/No):
  
 <details>
  <summary>additional information</summary>
  

  (_such as AKey [**NEVER EVER SHARE YOUR TOKEN!**], images_ or debug data from overview -> debug icon, your phone and android version)

</details>

\* **For very critical security breaches, please mail directly at info@evnotify.de**

GPlay97, 2020-01-13:
Bin mir nicht sicher, ob ich das Problem richtig verstanden habe.

Wenn man auf web.evnotify.de die ABRP Integration aktiviert und verlinkt, werden serverseitig alle übertragenen Daten auch an ABRP gesendet. Da entsteht also kein Mehrtraffic, da dies "im Hintergrund" vollautomatisiert passiert.

luckyheiko, 2020-01-14:
Dann scheint es wohl als hätte ich das falsch verstanden. 
EVNotivy sendet also immer die Daten an den Server?!

Das hatte ich so noch nicht gewusst und heißt für mich das ich die App nicht mehr nutze.

Vielen Dank für die Info

GPlay97, 2020-01-14:
?

Nur, wenn man die Option aktiviert, werden Daten übertragen, wie soll ABRP das denn sonst wissen?

luckyheiko, 2020-01-14:
Ist es angedachte mit ABRP und EVNotivy eine 'direkte' Verbindung herzustellen?
Also APP zu APP und nicht über den Webbrowser

Ok, dann ja immer noch meine 'Anfrage'. Ist die Übertragung Ohne Web angedacht um den Traffic zu vermeiden`?
was ja z.B auch gut wäre, wenn keine Handy Netz Verfügbar wäre.
(also kein Up / Download)

und P.S. ich habe nur 1GB im Monat und da muss ich bischen Haushalten und/oder andere Apps nutzen

GPlay97, 2020-01-14:
Ich glaube, ich verstehe die Frage nicht so recht. Wie soll denn etwas übertragen werden, ohne etwas hochzuladen / zu versenden?
Um eine Fernabfrage zu ermöglichen, braucht es Netz. Gibt es kein Netz, funktioniert EVNotify ja trotzdem, aber man kann es eben nicht von einem anderen Gerät, abgesehen von dem lokalen Gerät, einsehen.

EVNotify funktioniert auch mit langsamen, gedrosselten Internet

GPlay97, 2020-01-14:
Und wie zuvor geschrieben: Die ABRP Integration verbraucht keinen Mehrtraffic.

noradtux, 2020-01-14:
@luckyheiko ABRP erfordert immer eine Internetverbindung, die Datenübertragung läuft in jedem Fall über deren Backend. Die App zeigt nur die Daten vom Backend an.

maulwuff, 2020-09-12:
@luckyheiko folgendes Bild soll zeigen, wie die Daten fließen. 

Datenfluss mit beiden Apps:
```
[EvNotify-App] -- Daten -->   [EvNotifyServer] 
[ABRP-App]     <-- Daten --   [ABRP-Server]
```

Datenfluss mit beiden Apps und aktiver ABRP-Integration:
```
[EvNotify-App] -- Daten -->       [EvNotifyServer] -- Daten --> [ABRP-Server]
[ABRP-App]     <-- mehr Daten --  [ABRP-Server]
```
(Der EvNotifyServer sendet an ABRP, nicht das Handy)

Man sieht, das einzige was das Handy zusätzlich belastet, sind die auf das Handy eingehenden Live-Daten von ABRP. EvNotify belastet nicht zusätzlich.

closed 2022-07-30 by GPlay97