EVNotify/EVNotify #174: QRNotify wrong time remaining

eHarry87, 2020-01-13:
### New issue
- Short description: remaining time to full charge is to short
- Detailed description: when checking QR-code the showed time to full charge is way to short.

- Can you reproduce it? (_If so, how?_): Scan-QR-code

- [ ] critical issue? (_issue, which will affect multiple accounts, 
  an issue which prevents that you can use the app in a normal way,
  bug which makes app unusable, security breach*_) (Yes/No): No
  
 <details>
  <summary>additional information</summary>
  [web](https://drive.google.com/open?id=1aBllAX-iLvDNjP7zt9uGRK5T6g6Rj_sM)
  [QR](https://drive.google.com/open?id=1MJhyFSJ6Gdl0s8k33zX2POxF88W5opFj)
</details>

\* **For very critical security breaches, please mail directly at info@evnotify.de**

GPlay97, 2020-01-13:
Which car?

eHarry87, 2020-01-13:
Ioniq 28kWh EVNotiPi PlugnPlay

eHarry87, 2020-01-16:
[image]
[image]

charging speed is quite stable at 2.3kW

ektus, 2020-01-17:
The opposite may occur, too, if a charge limit is set in the vehicle (e.g. 80%) but EVNotify calculates the time to 100%. Is there a way to synchronise EVNotify with the target charge of the vehicle (in my case: e-Niro)?