EVNotify/EVNotify #175: No connection to car with 3 different ODB2 adapters

2xdehelft, 2020-01-20:
I’ve been trying to get EVNotify working on my 2019 Hyundai Kona 64KW. In the last couple of weeks I’ve ordered 3 different ODB2 adapters but with all have the same issue. 
After setting up the app with the correct ODB2 adapter I switch to the Dashboard tab. Leds on the ODB2 adapter start flashing, EVNotify tells me it is connecting but data doesn’t change. All values stay on 0. After some time EVNotify tells me no data is being received and its going to standby mode.
I’ve tested it with 2 different android phones (Galaxy S8 and Galaxy A30) but same behaviour. The same happens on both phones with all 3 ODB2 adapters.
I’ve tested the adapters with ELM327 identifier, 2 are ELM327 V1.5, one is a ELM327 V2.1. See link below for screenshots of the tests.
https://photos.app.goo.gl/MSjUopQDNt2w92td7

Ive tried another app for reading ODB2 error codes, and this actually works. It connects to ECU and gives me multiple sensor values, so the connection and adapters seem to work.

What more can I try to get EVNotify working? Is this just bad luck in ordering 3 different not-working ODB2 adapters?

GPlay97, 2020-01-27:
On dashboard page, while the phone interacts with the dongle - press the "debug" icon on the top please. After a few seconds (the led flashed a few times), click again to disable.

Then write AKey here or via mail (info@evnotify.de) to inspect what's going on.

2xdehelft, 2020-01-27:
> On dashboard page, while the phone interacts with the dongle - press the "debug" icon on the top please. After a few seconds (the led flashed a few times), click again to disable.
> 
> Then write AKey here or via mail ([info@evnotify.de](mailto:info@evnotify.de)) to inspect what's going on.

Thanks for your reply, just did the actions your requested. My akey is 6e1a28

GPlay97, 2020-01-27:
Thanks. Today and a few days ago there was data. Today the wrong car was selected, is this right?

2xdehelft, 2020-01-27:
No I have the right car selected, Hyundai Kona. I've also set the battery size (64kw). I am using the latest app now from the releases page, with capacity setting for battery. I've just ran another debug wedding from the app.

GPlay97, 2020-01-27:
Ah, now I see data. Maybe before the car was turned off?

However, there comes now data (and some days ago), but it's completely different unexpected data. So EVNotify itself acts correctly, but this might be a "bug". There is some special instruction needed for these kind of dongles.

I'll try to interpret the data

2xdehelft, 2020-01-27:
Yes the car was turned off but charging, now it was turned on. I didn't think it would matter...

Looks like it's a newer software in the car or something like that? It seems I am not the only one with this problem...

Again thanks for your time, it's a great car but really missing a decent app....

GPlay97, 2020-01-27:
Can you write the firmware of the car here?

2xdehelft, 2020-01-27:
Firmware:OSEV.EUR.287.180914.MICOM.DL
Software: OSEV.EUR.SOP.V100.190509.STD_M

That's both from the media system. Is that wat you need?

2xdehelft, 2020-01-31:
Have you been able to find anything? Is it caused by incompatible odb2 adapters or is something different about my car? Can you link a ODB2 adapter that works?

GPlay97, 2020-01-31:
Yes. The new upcoming update (maybe this weekend) should fix this for some dongle. It should work afterwards

2xdehelft, 2020-02-19:
Just to be sure, their has not yet been a new release right?

2xdehelft, 2020-05-12:
I just saw there will be a new release. Will this also be in this release? it wil be released in the play store>?

GPlay97, 2020-05-12:
@2xdehelft Yes, build works again.

Just had a quick look again at the debug data you sent a while ago.

This will be possible fixed with something already prepared, will test this asap.

Not included in the current upcoming release, but hopefully in the next one (that should also be available in the next days).

2xdehelft, 2020-05-12:
Awesome, thanks!

chosen21, 2020-05-27:
Hi,

I do have the same problem. With hyundai Ioniq 38kwh
Akey 9d9d0b
Software AE_E_PE.EUR.S5W_L.001.001.190630
Firmware AE_E_PE.EUR.01.190422.MICOM

GPlay97, 2020-05-27:
For IONIQ Facelift this is currently a bug, that it is not working. Hopefully I'm able to bring new update until end of this week.

2xdehelft, 2020-06-25:
> @2xdehelft Yes, build works again.
>
> Just had a quick look again at the debug data you sent a while ago.
>
> This will be possible fixed with something already prepared, will test this asap.
>
> Not [...]

Will this update ever be released?

GPlay97, 2020-06-25:
Yes, but currently I have several other high priorities regarding to EVNotify I'm working on.

AveNgelina, 2020-07-04:
That explains a lot, I was already desperate. Could that Info be added to the wiki please, to be found more easily? Thank you for your work =)