EVNotify/EVNotify #176: Can't connect to car first time (wrong date?????)

bluezetc, 2020-01-21:
### New issue
- Short description:  When trying to connect to the car I get the message:
"There has never been a connection to a car. Please connect your car first time (Invalid date)

- Detailed description:
I purchased this ODBII adaptor
https://www.amazon.co.uk/gp/product/B07PWWBYS8/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

which I thought was the same as the first one you recommended under compatibility
I can add it under odbII device but on returning to the dashboard screen I get the message above and no indication of what I should do. What does it mean "Invalid Date"

My car is a Niro EV

- Can you reproduce it? (_If so, how?_):

- [ ] critical issue? (_issue, which will affect multiple accounts, 
  an issue which prevents that you can use the app in a normal way,
  bug which makes app unusable, security breach*_) (Yes/No):
  
 <details>
  <summary>additional information</summary>
  

  (_such as AKey [**NEVER EVER SHARE YOUR TOKEN!**], images_ or debug data from overview -> debug icon, your phone and android version)

</details>

\* **For very critical security breaches, please mail directly at info@evnotify.de**

2xdehelft, 2020-01-27:
Same issue with my car (Kona) see https://github.com/EVNotify/EVNotify/issues/175

GPlay97, 2020-01-27:
https://github.com/EVNotify/EVNotify/issues/175#issuecomment-578690607

bluezetc, 2020-01-27:
> #175 (comment)

This Zircom adapter apparently works with Evnotify, but I ordered one and
got sent a generic ELM327 that didn't work - refund and trying to get
correct one.

https://www.amazon.co.uk/gp/product/B07DGBYPJV/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

Drumbud, 2021-06-26:
I have the vgate adapter obd2 blue tooth, and I am getting the same message. I have a 2021 Kona and a Samsung A20 phone running android 6