EVNotify/EVNotify #177: I can’t get a notification email or telegram

mreekster, 2020-01-24:
### New issue
- Short description: 
- Detailed description:

- Can you reproduce it? (_If so, how?_):

- [ ] critical issue? (_issue, which will affect multiple accounts, 
  an issue which prevents that you can use the app in a normal way,
  bug which makes app unusable, security breach*_) (Yes/No):
  
 <details>
  <summary>additional information</summary>
  

  (_such as AKey [**NEVER EVER SHARE YOUR TOKEN!**], images_ or debug data from overview -> debug icon, your phone and android version)

</details>

\* **For very critical security breaches, please mail directly at info@evnotify.de**

GPlay97, 2020-01-25:
Can you send me the AKey? (NOT Token) either here or via mail ? Info@evnotify.de

mreekster, 2020-01-26:
Emailed

konatpf, 2020-12-11:
I have the same problem

mreekster, 2020-12-11:
Trying to recall how I got it working. I have gotten it to work. Let me
check it out and see what I did to get it working

GPlay97, 2021-01-16:
For app it should work fine. If you still need help, don't hesitate to also contact info(at)EVNotify.de.

For PNP there was a big update (currently pre-release), that fixes unstable notifications: https://gplay97.me/posts/evnotify-big-pnp-update/

closed 2021-01-16 by GPlay97