EVNotify/EVNotify #178: Drive log export columns missing

MrApe, 2020-01-27:
### New issue
- Short description: exported log-files only contain SOC_DISPLAY, SOC_BMS and location

- Detailed description: The log file sent via email (Export.xlsx) only contains the columns "Date/Time", "State of charge (Display)" and "State of Charge (BMS)". All the other information from the apps log view (graphic plot) is missing like: DC_BATTERY_POWER, Temperatures and GPS_SPEED.

- Can you reproduce it?: Yes, just export a drive log.

- [ ] critical issue?: No

GPlay97, 2020-01-27:
This is known. The feature of exporting all data will be available soon

der-andyman, 2023-02-11:
Soooo, it has been 3 Years since this statement and yet, I still can't export my data. Only SOC (BMS) and SOC (Display) columns are exported... Still not possible to export everything? The log in the app shows all the needed information.

GPlay97, 2023-02-11:
In new backend for v3 it's planned - and even here it could be added. To be honest, I did not had this issue on my track list. But it's not motivating, if you write that here, give a beta feedback (with same content) and afterwards even send a 1 star public rating (again with same). It's open source. You can provide a pull request to contribute, donate or simply kindly ask again. You can also address the API directly to generate custom reports. 

Nevertheless, I will have a look at extending the export

der-andyman, 2023-02-11:
Sorry, I never worked with github before, I dont know what a pull request is...

Is it possible for you to look further into this please?? I am driving a Hyundai Kona and would love to have my data on my PC to monitor it further, but the exported logs only have the 2 said SOC colums.

iGadget, 2023-02-14:
@der-andyman perhaps you could consider [supporting](https://github.com/sponsors/EVNotify?frequency=recurring) the author's hard work? Might work a lot better than first bashing the guy and then asking for favors...

der-andyman, 2023-02-14:
I switches to another software. That topic is done for me

der-andyman, 2023-02-14:
> @der-andyman perhaps you could consider [supporting](https://github.com/sponsors/EVNotify?frequency=recurring) the author's hard work? Might work a lot better than first bashing the guy and then asking for favors...

And how did I "bash" him exactly? I just said that after 3 years (not exactly a short period of time) the issue is still there

GPlay97, 2023-02-14:
A lot happened in this time. Maybe not on this issue, but a lot of requests, features added, bugs fixed. A lightweight application of EVNotify (EVNotify Lite) was just released. Web Interface was developed . V3 is in development. I have limited time and limited financial budget, but still pay hundreds of euros every month.

I'm currently working more on EVNotify again and this is now also on my list. But with more and more users and data it is getting more and more challenging. Not only for financial server costs reasons, but also with the performance.

Exporting all data and logs "on demand" is therefore more a performance "issue" than a development issue. Since a few days I'm trying to fix some database issues, which slows me now down in developing "cool" stuff - but maintenance and support , emails, all that stuff needs to be done in the background. Not visible for you , I can understand that this is frustrating.

I wish I could spend more time, but at least in the past few weeks I was able again to get a few things done now :)

der-andyman, 2023-02-14:
@GPlay97 Thank you for this answer. I wish you best of luck with the development. If you felt attacked or smth by my initial answer, then I hereby apologize. I work in software development too and databases can be such a nightmare.