EVNotify/EVNotify #180: Wrong mail message: "You can drive 0km"

gertst, 2020-02-10:
### New issue
- I'v got a notification Email saying "The desired state of charge of 78.5% (Display) / 74.5% (BMS) has just been reached. With that, you can drive about 0km"
- Detailed description:
Looks like the nr of kms is zero.
I'm driving an e-Niro. 
The logs I got also have something strange: they are shown under "charges" instead of "drives".
EV-notify screenshots of overview and log: https://photos.app.goo.gl/QQyEjQCJB1Fbt4Et5
Android 10, Pixel 3

gertst, 2020-02-10:
Sorry, it looks like this issue was already reported earlier: https://github.com/EVNotify/EVNotify/issues/160

Don't know if there was a new release since then. I'm on v2.1.0:03.

I will test again by selecting car "Kone EV" and report back if this solves the issue.

GPlay97, 2020-02-10:
At least for the later issue, this has been reported. Fix for this in next update soon. For consumption - have you specified a consumption value within settings? But okay, 0 should normally lead to default value. So this seems to be a real bug

gertst, 2020-02-10:
Thanks for the quick response.
The consumption value is indeed set (to 18.5).

If you need more info, I'm happy to help.

Talking about the consumption: why is this a hard value from settings and not a calculation of the actual consumption? I assume this is only possible with quite some history data, so not that easy ...

gertst, 2020-02-11:
Here's an update with some more info:
I've changed from car profile: from Kia EV to Kona EV 
Indeed, now the mail shows the actual km: 

_The desired state of charge of 75.5% (Display) / 72% (BMS) has just been reached. With that, you can drive about 258km._

So the Kia EV profile is probably the cause of the issue? Because CEC and CED are missing?