EVNotify/EVNotify #181: OBD2 connection difficult while also connected to car: tip on how to resolve

gertst, 2020-02-11:
Connecting to the OBD2 device via bluetooth is not working very well on my e-Kona 

But I found out that, if the phone is not connected via Bluetooth to the car, it does work as expected. 

On my Kia e-Niro however, that Bluetooth connection to the car is made automatically, so it's a hassle to have to stop the car connection each time.

**A simple solution is to connect to Android Auto using an USB cable first, because then the bluetooth connection to the car is stopped, resulting in a working bluetooth connection to the OBD2 device ;-)**

My OBD2 device: [ScanTool OBDLink LX](https://www.amazon.com/gp/product/B00H9S71LW/ref=pe_2640190_232748420_pd_te_o_gr_ti?_encoding=UTF8&pd_rd_i=B00H9S71LW&pd_rd_r=1Y63CN3HTFDB3205GBPH&pd_rd_w=MpitG&pd_rd_wg=66JGu)

GPlay97, 2020-02-12:
What phone do you have? From my experience this depends also on the phone. Some phones have struggle to run multiple bluetooth connections at the same time

gertst, 2020-02-12:
I have a Google Pixel 3.
It's not always going wrong, but 3 out of 4 times it's not working. But like I said, I found a workaround ;-)