EVNotify/EVNotify #182: Kia e-Niro: "OBDII-error. Re-initialization" when car is not powered on.

gertst, 2020-02-11:
- Detailed description:
My car is switched on and there is a working connection between EVNotify and my OBD2 device.
Once I turn of the power of my car (pressing the Stop button), EVNotify shows a red notification at the bottom of the screen: "OBDII-error. Re-initialization. Please wait."
Then the blue bluetooth LED goes off for a second turns on again, the warning on EVNotify disappears for a few seconds, and all starts all over again. So the error keeps on showing up infinitely.

- Can you reproduce it? (_If so, how?_):
Yes, each time the car is turned off. Like explained above.

- [ ] critical issue? (_issue, which will affect multiple accounts, 
  an issue which prevents that you can use the app in a normal way,
  bug which makes app unusable, security breach*_) (Yes/No): 
Yes, I can not check charging my car without turning of my car.

GPlay97, 2020-02-12:
This is normal.

Once the car is off, no monitoring is technically possible, because no valid data can be retrieved via OBD. This is with every car.

It only works, when car is on or charging.

The re-initialization is, because EVNotify tries to re-establish a connection, because it could also be possible that car is on but due to some handshake error the communication was not possible and therefore need a re-init.

closed 2020-02-12 by GPlay97

gertst, 2020-02-12:
Ok, thanks for clarifying. 
But I think I got the error while charging. Not sure though - I will post back after double checking.

gertst, 2020-02-20:
I checked again and can confirm I had/have this error while charging, both when the e-Niro is powered on or off.
This issue is reproducible.
If you need more info I'm happy to help.

SebastianDietrich, 2020-07-11:
I cannot reproduce that issue on my car (KIA e-Niro 39) with motodia icar bluetooth dongle. While charging and power off it still gets data. When charging stops it no longer gets data.
In ev-notify I've set sync-modes forcieren (in hidden settings) and upload-only (the cloud symbol in the menu bar)

gertst, 2020-07-18:
Here's a workaround than might be useful for others:

I still have the initial issue, probably due to my ODB2 dongle ([Konnwei](https://www.amazon.com/gp/product/B07X6286CR/ref=ppx_od_dt_b_asin_title_s00?ie=UTF8&psc=1)).

But I managed to bypass it:
- connect to the dongle while the car is on
- turn the car off - the connection should stay
- important: don't connect to Google Auto, as this seems to stops the ODB2 connection sometimes once the car is turned off.
- I also set "force sync modes" to true in EVNotify (hidden settings), like @SebastianDietrich suggested.

mhantsch, 2020-07-27:
I found that on some Androids (depend on phone manufacturer), EVNotify gets killed or at least deactivated by the OS (I guess in order to save battery power). For example, on OnePlus OxygenOS I set the battery management for EVNotfy to "do not optimize", AND also I had to lock EVNotify in the recent apps list, in order to prevent OxygenOS from killing it.

On a OnePlus One with LineageOS 17.1 , I needed to have an Automate script that kept "activating" EVNotify by sending it to the foreground every 10 seconds or so. That was the only way I found to keep it active even during charging.