EVNotify/EVNotify #184: Support for MG ZS EV

VinceW31, 2020-02-17:
Request

Support for MG ZS EV

noradtux, 2020-02-17:
Can you point us to any MG ZS EV CAN-Bus documentation?

VinceW31, 2020-02-17:
Hi, have a look here: https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/issues/252

speculatrix, 2021-01-16:
Alternatively, could you consider having an OVMS mode where the app talks to the ovms server instead?

GPlay97, 2021-01-16:
Currently we are implementing a few new cars and this might be one of the next ones, stay tuned and patient

meltymcface, 2021-12-06:
Hi! Is MG ZS support anywhere closer? Many thanks :)

ttjaden, 2022-04-21:
Hi! I also want to ask if the MG ZS EV Comfort (Facelift) Model will be available? I'm contributing and sponsoring the [EVCC](https://github.com/evcc-io/evcc) Project. An to integrate my MG ZS EV into EVCC i want to use EVNotify.

Is there something I could help with testing or so on?

OzGav, 2022-10-10:
Fingers crossed the V3 that is being worked on supported the MG. Although this guys said he was working on support for the current version but that doesn't seem to have gone anywhere https://www.mgevs.com/threads/evnotify-monitor-your-electric-vehicle-remotely.2239/