EVNotify/EVNotify #186: Support for NISSAN LEAF

thanarious, 2020-02-26:
Did an extensive search in the homepage, twitter accounts and in the repository here, and didn't see any requests or questions regarding EVNotify supporting NISSAN LEAFs. So I thought I'd ask. Since there's already LeafSPY that supports the LEAF (although not open source afaik) and there's also OVMS (which is an open source project), wouldn't it be relatively easy to add LEAF support to EVNotify?

Let me know if there's any way to help by providing more information, since I have a working ODBII WiFi connection between my LEAF and iOS/android.

Malikster, 2020-09-30:
Hi. Any updates on this? I am interested too.

TweetsOfNiklas, 2021-03-20:
Since Nissan Connect Service stays being unreliable it would be great to have EVNotify to monitor the battery status. Also it should be possible to start AC via the app.

antonhagg, 2021-06-17:
Would be nice with Nissan Leaf support. Sign me up for bug testing!

jatgm1, 2022-02-04:
Ya I am really trying to get something to access my my 2012 which carwings was discontinued on. That would be absolutely amazing