EVNotify/EVNotify #187: Vgate iCar Pro BT 4.0

SappiBT, 2020-03-07:
Bought the Vgate OBD2 dongle.

Works OK, connects to my iPhone (iOS 13) powers off (-30 min) after stopping te car (Hyundai Kona Electric), but never powers up when I start the car again. Specs say it does, but not here. Have to take it out and in again before it works again.

Anybode?

SliVee, 2020-04-20:
I have same dongle and it powers up just fine when I start he car and I have also Kona EV. I just need to figure out how EVNotify can connect automatically to dongle. It takes usually time and multiple tries to connect.

lepinkainen, 2020-09-28:
Hyundais (at least my Ioniq) power down the OBD bus after 30 minutes. It does stay on if you are charging though.

mhantsch, 2020-10-07:
@SliVee EVNotify will try to reconnect to the Dongle if you tap on the Settings button, then on Dashboard button (the buttons on the bottom of the screen). I made an 'Automate' flow for my phone to do that automatically whenever the phone in the car gets power.

philos31, 2020-10-08:
@mhantsch interesting, could you share that automate flow? I seem to have the same problem that evnotify does not autoconnect to the dongle. So I would be interested in learning how to automate that. Thanks (Danke)

mhantsch, 2020-10-15:
@philos31 check this out: https://www.goingelectric.de/wiki/EVNotify-Alternative-Loesung-mit-Automate/
(In German, but hopefully Google Translate can help you.)

Tchakpoum, 2020-11-19:
> 
> 
> Bought the Vgate OBD2 dongle.
> 
> Works OK, connects to my iPhone (iOS 13) powers off (-30 min) after stopping te car (Hyundai Kona Electric), but never powers up when I start the car again. Specs say it does, but not here. Have to take it out and in again before it works again.
> 
> Anybode?

Same for me !

i can configure the dongle whith "Autodoctor" but cant link him on EVnotify and after the dongle shut down, he didnt start again...

I am using IOS and i think its a part of the problem :-/

GPlay97, 2020-11-19:
Yes. Known issue on iOS. Will get better hopefully with v3. Currently the main reason is also a restriction from Apple since it's unfortunately forbidden for me to run as background task

Tchakpoum, 2020-11-19:
> 
> 
> Yes. Known issue on iOS. Will get better hopefully with v3. Currently the main reason is also a restriction from Apple since it's unfortunately forbidden for me to run as background task

Apple or the "Downgrade realm" !

when did the V3 is planned ?

Thanks for all btw !