EVNotify/EVNotify #189: Stopped using EvNotify

SappiBT, 2020-03-21:
I stopped using ABRP and EvNotify because of the instability of both app's and even more in combination with a OBD2 dongle.

The concept is great, the possibility's also, but the soft- and hardware is not ready yet for daily use.

Sorry!

GPlay97, 2020-03-21:
Sorry to hear that.

Can you give more details?

This really much depends on which Dongle you use. Unfortunately the most reliable version is and will always be the "PlugAndPlay" device, that can be bought, because this uses a standardized obd interface.

Besides that, if you have a good obd 2 Dongle, it works really great in daily usage.
What exactly was your issue?

The sync to ABRP relies on them. Unfortunately they often have issues, especially with their endpoints and reliability as well as availability. That's true.

SappiBT, 2020-03-22:
I used the Vgate Icar Pro BT 4.0 dongle (as advised) en tried also 3 others.

GPlay97, 2020-03-22:
On which platform? Android or iOS?

SappiBT, 2020-03-22:
iOS

GPlay97, 2020-03-22:
Well, this explains everything.

The iOS app has two disadvantages, it has a bug, where it crashes sometimes.

But the biggest issue is that it can not run in background because apple declines it. Arguing with apple is so hard you can not image this. That's why there is no way for me to allow reliable background communication like it exists in Android app. Furthermore apple does not support classic Bluetooth protocol which is also the reason why not every Dongle is working.

Android app is million times better. PlugAndPlay the best.

It is unknown, when these issues will be fixed or if they can be fixed without a complete standalone app only for iOS..

SappiBT, 2020-03-22:
OK. To bad. Thanks.
 

George Tabak
tabag55@xs4all.nl

green-us, 2020-04-12:
Hi Gplay,

You're referring to the "the "PlugAndPlay" device" . Is that still available somewhere? And furthermore I see a decrease in maintenance on GitHub: Are you all well and/or can you give us an update/insight, maybe in a new thread on where EVnotify is heading? Thanks a lot (wo)man!

GPlay97, 2020-04-12:
Hey @green-us ,

yes, it's available to buy via shop.evnotify.de (currently available only in German, but it will be available soon in English too - if you have questions or want to buy, contact info@evnotify.de).

Everything is fine, thanks for asking.

However, I'm currently rewriting / refactoring a lot, starting with the server currently (you can see single services being created under EVNotify in different repositories) - I'm also planning and developing currently a complete new, native App.
That's why you currently see here no progress, but a lot of stuff happens in background :)

More insights will follow :)

green-us, 2020-04-14:
Thanks for the update @GPlay97! I'm curious to what you will come up with next.

edword99, 2021-05-26:
I recently got the iOS version working as long as I leave the app in the foreground, set the display to never sleep (via phone settings) and use a vgate iCar Pro BLE Dongle. Unfortunately it is still suffering from a bug that causes it to crash every few minutes. Is there anything we can do to stop the crash and/or is their any update on getting this bug fixed? Thanks for making this for iOS!

GPlay97, 2021-05-26:
I'm working on a new update and will also try to migrate to newer dependencies, this might fix issues. Besides that, the new app I'm working on in the background is built on top of directly native features and should be way more reliable then. I'll keep you posted

edword99, 2021-05-26:
That’s great to hear! Thanks for the update @GPlay97

burnsfisher, 2021-06-09:
Just a comment that the trick described by edword99 works well for me as well.  Looking forward to the new version!