EVNotify/EVNotify #191: Any tips for (auto) connecting OBD2 to EVNotify?

SliVee, 2020-04-20:
Hi!

Thanks for the app and hard work with it. I have difficulties to connect Samsung Galaxy S4 (Android 5) and S10 (Android 10) to EVNotify using Vgate iCar Pro Bluetooth 4.0. Car is Kona EV 39 kWh. I use now mainly Galaxy S4 and phone sits on my car with powerbank. Only BT connection is set to dongle. I usually have to restart the app or try to go to log entries and back to Dashboard to get "Initializing.." text pop up. After ~4-5 tries it connects and gives data. I know Vgate might not be the best OBD2 scanner so maybe it is just slow?

I was wondering what are the best settings to get auto connection OBD2 to EVNofity? If it is even possible. The dream would be leave phone on screen off and EVNotify would start updating data automatically to server when I start the car and start to drive. I've tried to leave EVNotify app running (just close the screen) but it seems I have to unlock phone before it connects again. OBD2 gets power when I start the car (red led). Bluetooth is on 24/7.

I've also tried some hidden settings like AUTO_WAKEUP and BACKGROUND_MODE. I couldn't find anywhere on Wiki what these settings do exactly. Is anywhere list what these hidden settings do?

EVSalomon, 2020-04-20:
I've rooted my Galaxy S4 mini and installed the App "Automate".
There i have made a sketch that is setting my phone in flight mode and close the app if the phone gets no more power from the USB port. If it gets power again, the sketch restarts the app and deactivates the flight mode. The USB-Port next to the QI-Pad gets also Power while charging. In flight mode, my phone lasts for several days.

SliVee, 2020-04-20:
@EVSalomon that sounds great! Do you get fast connection OBD2 to EVNotify with this setup?

EVSalomon, 2020-04-21:
i think the obd connects in like 30s after starting the car.

GPlay97, 2020-04-21:
Thank you @EVSalomon for sharing your solution.

Background_mode + persistent notification is to prevent battery optimizations, so app runs reliable in background, when display lock is on, once connection is established.
However to prevent 12v battery drain, after car is turned off / no valid data is received, it goes to deep sleep. This behavior was planned to be overriden by auto_wakeup, however it was not implemented, since the software solution to automate it, was not reliable for all obd dongles, so it could still happen, that the 12v battery was empty.

Depending on the obd dongle, it takes a while / a few attempts, until valid data can be retrieved, unfortunately every dongle is different.

The only really reliable and automated way currently is the PlugAndPlay device, since it has a standardized chip as well as hardware 12v watcher. It can be bought via https://shop.evnotify.de - it will be available soon in English and shippable within EU.

SliVee, 2020-05-14:
Can recommend @EVSalomon solution. Screen can be locked with PIN code and it still connects automatically without unlocking phone. It feels like if you start the car and stay while in P (parking) mode it doesn't connect or at least give information. I configured my automate that it waits 30 seconds before it launches EVNotify app so I have plenty of time to start driving and data will flow.

This isn't 100% working and sometimes EVNotify just can't get connection but most of the times it connects. It seems to help if you boot the whole Galaxy S4 to get it working again.

@GPlay97 thanks for explaining some of to those hidden settings. Your PlugAndPlay device would be neat but I will keep this Kona about 1 year and then I'm looking something with bigger battery (Skoda Enyaq 62/82 or Kona EV 64).

mhantsch, 2020-06-01:
> However to prevent 12v battery drain, after car is turned off / no valid data is received, it goes to deep sleep. This behavior was planned to be overriden by auto_wakeup, however it was not [...]

I still think there should be an automatic wakeup in EVnotify; users can choose to enable it or not.

- As long as device power is off, there cannot be any wakeup.
- When EVnotify was in deep sleep, and power is turned on, it should try to reconnect to the dongle. As long as power is on, try reconnect for up to a configurable number of retries (e.g. 10 times), with a timeout of a configurable number of seconds on each attempt (e.g. 30 seconds). If it cannot reestablish the connection after all retries, go back to sleep until the next time power is turned on.
- If device power is turned off during the reconnection attempts, stop retrying and go back to sleep.

I really hope the EVnotify rewrite will have this functionality. It would solve a lot of hassles with our attempts to script around this with Automate or Tasker.

WebSpider, 2021-09-02:
Hey all!

I tried to set this up on my Skoda Enyaq iV 80, but there is a caveat that really should be taken into account:

Modern car alarms activate when the OBD2 port is used under suspicious circumstances, in order to prevent key cloning.
One of those circumstances is when the car is locked, and the windows closed.

Letting EVNotify connect at the wrong moment, would trigger my car alarm (and wake up my neighbors :grin: )

sorentorp, 2022-01-03:
@WebSpider did you get it to work? I'm almost there. But for some reason my Android phone doesn't save the Pinkode for the Vgate icar2 dongle, so I have to type it manually every time.
Does anyone know how to save the pincode?