EVNotify/EVNotify #197: Crashes on Android after a while

p3ntium, 2020-05-16:
### New issue
- Short description: The EVNotify crash on Android after few minutes
- Detailed description and reproduce: When I start the vehicle charge, I starts the EVnotify on my Samsung A3 and it works well (link with the dongle and send the data to the cloud), but when a few minutes has been passed (around 5), the EVNotify goes down and I need to restart it.

Any ideas?

GPlay97, 2020-05-16:
Can you post the exact Android versions and patches here from system info?

p3ntium, 2020-05-16:
Yes:

Android 9
Base band: J330FNXXU3CSK2
kernel versión 3.18.91-17280703
Security level patch for Android: 1 nov 2019

I've only tested in EVNotify 2.1.0, but I think with the v2.2.0 the same happens, because in the changelog there is no comments about crash fixes...

Thanks!

GPlay97, 2020-05-16:
In my code I think it's unlikely.

However it could be a bug in the Bluetooth plugin EVNotify uses.

I will have a further look into it.

On iOS it often crashes, that's a known issue. On Android it worked actually quite stable

p3ntium, 2020-05-16:
Ok, now I'm installing a new updated of the OS (after I check new updates), and then I will install the 2.2.0 of EVNotify and will try again.
I will feedback to you after test completed.
Regards!

GPlay97, 2020-05-16:
Thanks.

Maybe uninstall EVNotify also and install it again to have a fresh installation. Do you have auto awake or keep alive within EVNotify (hidden) settings enabled? Maybe this could also help (or produce the issue, depends ;) )

GPlay97, 2020-12-11:
Closed due to no new response and information.

closed 2020-12-11 by GPlay97