EVNotify/EVNotify #199: Unable to build Android app

bje-, 2020-06-01:
### New issue
- Short description: 
Unable to build the app with Cordova.

- Detailed description:

I'm unable to build following the instructions in `CONTRIBUTIONS.md`:

```
~/source/EVNotify (master)$ node --version
v10.19.0
~/source/EVNotify (master)$ npm --version
6.14.4
~/source/EVNotify (master)$ cordova --version
9.0.0 (cordova-lib@9.0.1)
~/source/EVNotify/app (master)$ cordova platform add android
Using cordova-fetch for cordova-android@^8.1.0
~/source/EVNotify/app (master)$ cordova build android
cordova-plugin-fcm: You have installed platform android but file 'google-services.json' was not found in your Cordova project root folder.
```

ektus, 2020-06-02:
See https://github.com/EVNotify/EVNotify/issues/159#issuecomment-602609027

GPlay97, 2020-06-02:
As written on other issue, you would need a valid dev account for Firebase.

The plugin is used for sending realtime notifications.

You can either create your own file - or just drop the plugin, it will still build

closed 2020-12-11 by GPlay97

joopdo, 2022-01-21:
@GPlay97 Thanks for your hard work. If possible, could you share some pointers about your current build environment? (Java, Cordova version)

joopdo, 2022-02-08:
Managed to build with following:

Cordova 11
Gradle 5.6.3

```
git clone https://github.com/GPlay97/EVNotify
cp <path>/EVNotify/app/client.js ./
cordova platform add android@8.0.0
cordova plugin rm cordova-plugin-fcm

```

Not that it works, EVNotify opens a white screen, logcat shows:

```
02-08 11:41:06.723  9293  9293 D CordovaActivity: Resumed the activity.
02-08 11:41:06.803  1718  1842 I ActivityTaskManager: Displayed com.evnotify.app/.MainActivity: +337ms
02-08 11:41:06.817  9293  9293 D CordovaWebViewImpl: onPageDidNavigate(file:///android_asset/www/index.html)
02-08 11:41:06.837  9293  9322 E AndroidProtocolHandler: Unable to open asset URL: file:///android_asset/www/client.js
02-08 11:41:06.843  9293  9322 E AndroidProtocolHandler: Unable to open asset URL: file:///android_asset/www/js/bundle.js
02-08 11:41:07.109  9293  9293 D JsMessageQueue: Set native->JS mode to EvalBridgeMode
02-08 11:41:07.112  9293  9293 D CordovaWebViewImpl: onPageFinished(file:///android_asset/www/index.html)
02-08 11:41:07.113  9293  9370 E AndroidProtocolHandler: Unable to open asset URL: file:///android_asset/www/cordova_plugins.js
```