EVNotify/EVNotify #200: Chevy Bolt / Opel Ampera-e: full support not there yet

wypiki, 2020-06-05:
Full support shouldn't take a long time to implement because the PIDs are widely known. According to ABRP:
!Air Temp 2
!Batt Temp
!Battery Capacity (2017-2018 only)
!Battery Level Displayed
!Charger HV Power
!HV Current
!Inst Kpower
!MG Voltage
from this list: http://bit.ly/ChevyBoltPIDs

Please add the necessary PIDs. I would be willing to test it also.

noradtux, 2020-06-05:
Feel free to provide code :)

BenMueller, 2021-05-05:
Subscribing to this topic: I don't have the time now but I can look into writing some code for the Bolt once I get a ODB2 dongle and have free time (currently moving cross-country and finishing finals so things are hectic)