EVNotify/EVNotify #201: EV-Notify web API documentation

SebastianDietrich, 2020-06-16:
Is there anywhere a documentation for the web-api (https://app.evnotify.de/soc?akey=<my-akey>&token=<my-token>)
I get e.g. {"soc_display":55.5,"soc_bms":null,"last_soc":1592317780} and assume that:
* 55.5 ist the soc percentage as stated in https://web.evnotify.de/#/
* last_soc is the unix-time of the last connection (1592317780 = 2020-06-16T14:29:40+00:00)

but I don't know which values to expect from bms (battery-management-system ?)
And especially I'd like to know how frequently I might access the API (e.g. every 10 sec???)

joopdo, 2022-01-20:
Also to include https://app.evnotify.de/extended?akey=...

iGadget, 2022-09-27:
> For others finding this -> https://api.evnotify.de/
> 
> Also to include https://app.evnotify.de/extended?akey=...

I get a security warning that the certificate belongs to mail.spz.su... What's going on here?

GPlay97, 2022-09-29:
The subdomain no longer exists, since the api documentation of v2 is no longer available because I'm focusing now v3. New testers will be invited for the new api. The current api still works. New documentation will be generated soon for the new experimental v3 api. I just deleted the subdomain for now, otherwise it was managed by the hosting provider (why you got the security warning, since the existing server for api docs no longer exists)