EVNotify/EVNotify #203: Information on account privacy policy

CopyCat73, 2020-07-13:
Not exactly a technical issue, but I cannot find much information in the wiki or on the site why an account is needed and what kind of information you store (in relation to GDPR) and for what purpose.

solexx, 2020-09-29:
ACK. Actually I had trouble understanding how everything works until it dawned on me that all information that the app displays is actually stored remotely. I skimmed the t&c and privacy policies on https://shop.evnotify.de/ and even that appears to be incomplete in that regard. I know these things are a pain, especially if you do this stuff non-commercially in your free time. But you are asking for legal trouble if you are not clear about where user data is actually stored. I also assume that @GPlay97 has direct access to this data (probably including GPS). Personally, I do not care about that in this specific instance, but others might see this differently.

GPlay97, 2020-09-29:
Thanks for the hint, I'll add more information very soon to clarify the things.