EVNotify/EVNotify #204: Unlink or reset abrp integration

cmunk, 2020-07-20:
### New issue
- Short description: cannot reset / unlink abrp integration 
- Detailed description: I use my OBD2 between devices but my evnotify account is linked to one specific abrp car. So I would like to unlink to link the other car instead 

Thanjs

dimzendk, 2021-01-16:
If reset is needed, do we have to get you to do it at this time?

GPlay97, 2021-01-16:
@dimzendk Yes, exactly. I'll then remove the linked token manually. I'll contact ABRP to provide us an API for future versions to also remove it on their side (that's why it's not "officially" possible, since the removal is only on our side currently).

dimzendk, 2021-01-16:
> @dimzendk Yes, exactly. I'll then remove the linked token manually. I'll contact ABRP to provide us an API for future versions to also remove it on their side (that's why it's not "officially" possible, since the removal is only on our side currently).

I could unlink on ABRP's page, so it's only on EVnotify I need it unlinked.
What and where do you need that info? :)

GPlay97, 2021-01-16:
Thank you for the info - I'll ask for the API to do that then also on our side to them.

For current removal: Write me a mail with your AKey to info(at)EVNotify.de

patriknoomi, 2021-05-20:
I would also need this feature. I accidently connected EV notify to the wrong ABRP account... and now can't get it to work.

GPlay97, 2021-09-26:
Can now be done on web interface web.evnotify.de - will be added to app too, once feedback is positive for unlink feature.