EVNotify/EVNotify #205: Opel Corsa-e support

TheHawk1337, 2020-09-09:
New Opel Corsa-e is not supported, failure when trying to connect to the car. Note that this car uses the same platform as the e-208, e-2008, e-Mokka and probaly more to come. Since it's from PSA instead of GM the Ampera-e car setting also fails.

I got the car, let me know how I can help. I have *some* programming experience but not with Android.

Thanks

just-grizzle, 2020-10-14:
I have a BT OBD dongle and am willing to support that implenentation, too. But I have no clue where to start to find the right tools, PIDs, ...
Someone could point to the right direction?

ysamjo, 2020-10-21:
More or less same here. Got the car and would like to see it supported.

What's the underlying issue? That PSA doesn't want it or simply ressources?

noradtux, 2020-10-21:
Resources. Has anyone reverse engineered the can bus yet? That is a requisite for the possibility of any support.

just-grizzle, 2020-11-03:
I would try to, if I knew where to start. Maybe this question already disqualifies me as a helper. 🤓 
Nevertheless, if there's any chance I can support, please let me know.

AndreaL77, 2020-12-18:
Hi

I have a corsa-e too and I would like to help.
I am some skills in linux, iot.

I saw that PSA offers api access but currently is still not open to pubblic use
https://developer.groupe-psa.io/

Joseph-Muc, 2020-12-22:
New Opel Corsa-e is not supported...........................! Me too.

wrecsamwales, 2020-12-31:
Another me too, for the Vauxhall Corsa-e please.

noradtux, 2021-01-13:
As said, to make that possible we either need a description of how to talk to the car or we need the car plus maybe the official diagnostic device to reverse engineer that stuff.

TheHawk1337, 2021-01-19:
I emailed Opel asking for their help with an app like this. I don't expect much but who knows.

If that doesn't work out I might be able to do something via the local dealer.
In case I do it via the local Opel dealer, what would be a good approach? Use some sort of odb sniffer when they plug in their diagnostics? I might be able to do something like that using a splitter + some python scripts with python-obd if I can get that to work.

Perhaps I could also brute-force something via the script but that's not my favorite option :)

I'll ask my dealer quickly anyway today since I'll be picking up the car from an airbag recall later today anyway.

Joseph-Muc, 2021-01-19:
My trial results are not very satisfactroy, and not successful. Perhaps they are usefull anyway:
Equipment 1:
OBD to Bluetooth Dongle (vgate - ELM327 Version 2.2)
Android - APP: OBD Now Terminal Version 1.0.16, Android Version 10
Result: Autodetect Protokol: ISO 15765-4 CAN (11 bit ID, 500 kbaud)
            Send Broadcast: hex: 7df 02 01 00  > No Data (expected: Bitmap of PIDs supported)
            Command ATMA:      Bufferoverflow after some data and a lot of "DATA ERROR"
Equipment 2:
USB-to-CAN

noradtux, 2021-01-19:
In general sniffing traffic of an official diagnostic device is a good approach. But anything ELM327-based (BT or USB) will be way to slow for that. What works great for me is a Raspberry Pi with a MCP2515 based CAN-hat (like the one from Waveshare). Those can actually deal with the 500kBit coming from the car.

gotgot1995, 2021-03-30:
I drive a Peugeot e-208 (which is the same as the Opel Corsa-e and the rest of the PSA group), I've got a RaspberryPi and I might get my hands on an official Peugeot diagnostic device.

CarstenWa, 2021-04-17:
They are using a special adapter cable for it.

hlinden, 2021-04-19:
Here's some images of the adapter:

[image]
[image]
[image]

SamuelBrucksch, 2021-04-22:
Here is a pretty good description of the commands that need to be sent to get data:
https://www.goingelectric.de/forum/viewtopic.php?p=1514148#p1514148

```
#######
####### Setup commands
#######
ATWS ---> Warm reset set back to defaults
ATI ---> Show interface ID
ATE0 ---> Turn off terminal echo
AT@1 --->
ATSP6 ---> Set to protocol 6: ISO 15765-4 CAN (11 bit ID 500 kbaud)
ATAT1 --->
ATL0 ---> Disable line feed
ATS0 ---> Disable sending spaces between hex characters
ATH1 ---> Turn on odb2 header
ATCAF1 ---> Turn on CAN auto formatting
ATSH79B ---> Set request message header 79B
ATFCSH79B ---> Set flow control header (CAN ID)
ATFCSD300000 ---> Set flow control data
ATFCSM1 ---> Set flow control mode; 1 = User defined CAN ID and data

#######
####### Data commands
#######
ATSH6A2 ---> Set the header of transmitted OBD messages to header 6A2
ATFCSH6A2 ---> Set flow control header (CAN ID)
ATCRA682 ---> Set the CAN hardware filter to 682
22D4341 ---> Ambient temperature
22D8EF1 ---> Battery temperature
22D4021 ---> Speed
ATSH6B4 ---> Set the header of transmitted OBD messages to header 6B4
ATFCSH6B4 ---> Set flow control header (CAN ID)
ATCRA694 ---> Set the CAN hardware filter to 694
22D8161 ---> Battery current
22D8151 ---> Battery voltage
22D86F1 ---> Battery minimum voltage
22D8701 ---> Battery maximum voltage
22D4101 ---> SOC calibrated
22D8651 ---> kWh available
22D8601 ---> SOH
22D8101 ---> SOC
```

And the already decoded values:
SoC calibrated (22D4101) = value / 512
bat min (22D86F1 ) = value in mV
mat max (22D8701) = value in mV

So it looks like it's easy to get the data but not all of them are decoded yet.

I just looked at one of the car files, lets just take the ioniq:
https://github.com/EVNotify/EVNotify/blob/f87449c7fdf4fe868c48c0bec13a5df62b8f22b5/app/www/components/cars/IONIQ_BEV.vue#L9-L17
If i take one of those car files as base to get data from one of the PSA cars, what should i enter in that initialization section to get the data from the above mentioned values? If anyone can help me setup a basic car file for PSA cars i can try to decode some more values.

GPlay97, 2021-04-22:
Nice @SamuelBrucksch 

Can you send me a mail to info(at) EVNotify.de

I'll create a test file for you

SamuelBrucksch, 2021-04-22:
Sent you a mail. Are there instructions somewhere how to build the app and deploy it to android so i can adjust and test myself?

GPlay97, 2021-04-23:
Yes, indeed the setup instructions needs to be updated. Since EVNotify will soon start to be native, this current codebase will be obsolete. The Google services json is from firebase console that you can get with your own account.

CarstenWa, 2021-04-24:
If you need some help to retrieve data.. Will get mine corsa in about 5 or 6 days.

CarstenWa, 2021-05-02:
Anything new?

gotgot1995, 2021-06-30:
I got my hands on an official diagnostic tool and I could retrieve all information about the BMS and more from the CAN bus using the PSA-Group diagnostic software (running on Windows). I guess I could try to spoof the CAN bus to retrieve some data but I just don't know how to proceed. Any ideas on how to do this right ?

noradtux, 2021-06-30:
Best (and least expensive) way to sniff CAN-data I know is to use a 
Raspberry Pi with an MCP2515-based CAN-hat (like the one from 
Waveshare). Use a Y-cable and sniff all communication between the car 
and the diagnostic device. ELM327 and clones tend to be too slow for 
proper sniffing.

moritz31, 2021-08-04:
@gotgot1995 do you got any results, with the help of the Offical diagnostik?

gotgot1995, 2021-08-09:
Hi @moritz31, I'm working on it as we speak. 😄 

Using the official diagnostic tool and software I was able to retrieve useful information such as SoC, instantaneous power output and battery temperature. I'm using a [Microchip CAN bus analyser](https://ww1.microchip.com/downloads/en/DeviceDoc/51848B.pdf) to retrieve the dataframes. I should have them by the end of the week.

My only concern is that the PSA-Group BEVs use connectors 3 and 8 (instead of 6 and 14 as standard) so I'm hoping that some OBD dongle on the market can capture data on these connectors. Otherwise people are going to have to build or buy [that kind of adapter](https://www.amazon.fr/Power-Cruise-Control-Adaptateur-c%C3%A2ble/dp/B08MXS8W3C/ref=sr_1_5?dchild=1&keywords=Power+Cruise+Control+PSA+adapter&link_code=qs&qid=1628501517&sourceid=Mozilla-search&sr=8-5).

SamuelBrucksch, 2021-08-09:
Latest infotainment update brings OBD pins to default pins. So no need for cable anymore.

gotgot1995, 2021-08-09:
@SamuelBrucksch really? Well that would be very good news. 👍

I went to my Peugeot dealer recently and they have updated my BMS firmware. So maybe they also update that part. In any case, I will be able to confirm the connectors shortly.

CarstenWa, 2021-08-09:
It works also without the update. Tested it few hours ago. It seems the only thing that the adapters do is to switch off the dongle if you power down the car.

SamuelBrucksch, 2021-08-09:
It definitely did not work before on many cars. Only after a certain update or production date (where i assume that update is there already) it works. I checked both and suddenly after infotainment update it worked for my car.

CarstenWa, 2021-08-09:
Mine was produced February 2021. Will see what the update will update. 7.6G for a Touchscreen with SatNav hmmmm

gotgot1995, 2021-08-09:
@SamuelBrucksch okay, nice. Which one is your car again? Opel corsa-e?

SamuelBrucksch, 2021-08-09:
Zafira-e, but as i wrote my own OBD app ABOR i also got feedback from Corsa users.

CarstenWa, 2021-08-09:
> Zafira-e, but as i wrote my own OBD app ABOR i also got feedback from Corsa users.

Ragnarid?

gotgot1995, 2021-08-09:
> Zafira-e, but as i wrote my own OBD app ABOR i also got feedback from Corsa users.

Awesome. 👍 It's good to hear that the community is moving forward on the subject.

gotgot1995, 2021-08-09:
> It definitely did not work before on many cars. Only after a certain update or production date (where i assume that update is there already) it works. I checked both and suddenly after infotainment update it worked for my car. 
> 
> The PCC adapter changes the pins and uses the default always on power source, but you can reorder them to use the only power on when ignition on.
> 
> The "old" pins also still work. As i built my own adapter i can directly compare it and it works on old and new pins now which it did not before.
> 
> BTW my car does not provide OBD values when not powered, even when charging no values come (I have a Zafira-e though but it should be same on other PSA cars as PIDs and so on are also the same, so maybe same firmware for that behind).

Good evening @SamuelBrucksch, thanks again for all this precious detail. I tried to make my own Y cable and spoof connectors 6 and 14 (the ones you call "the old pins" correct?) and I failed completely even though I had tested my cable before. 😅

When I powered on my e-208, the instrument cluster went crazy with tons of fatal errors! 😰 I had to turn everything off and unplug everything. Luckily when I turned it back on, all the errors were gone the car is running just fine. 😪

I'll investigate tomorrow but I think I connected the Microchip analyser badly and caused a short-circuit. Maybe I should use the PCC adapter instead of trying to make my own cables... 😔

gotgot1995, 2021-08-10:
I finally managed to capture all traffic going through the CAN bus using connectors 6 and 14.

I started "simple" and requested the following parameters to the BMS using the official diagnostic tool & software:
- **calibrated SoC:** 70.1%
- **SoC:** 70.1%
- **maximum SoC:** 70.0%
- **minimum SoC:** 68.7%
- **average Temperature:** 22°C
- **max Temperature:** 23°C
- **min Temperature:** 21°C

@GPlay97 @noradtux Please, DM me if you're interested to get the raw capture file for reverse-engineering purposes. :wink:

Zack-95, 2021-08-31:
Has anyone figured out how the current frame is parsed to get the right data ?
22D8161 ---> Battery current
I have this exemple of a frame :
694 8 7 62 D8 16 0 1 2C 1 
but I can't figure how to get the current value from this Hex frame (0 1 2C 1 )

> Here is a pretty good description of the commands that need to be sent to get data:
> https://www.goingelectric.de/forum/viewtopic.php?p=1514148#p1514148
> 
> ```
> #######
> ####### Setup commands
> #######
> ATWS ---> Warm reset set back to defaults
> ATI ---> Show interface ID
> ATE0 ---> Turn off terminal echo
> AT@1 --->
> ATSP6 ---> Set to protocol 6: ISO 15765-4 CAN (11 bit ID 500 kbaud)
> ATAT1 --->
> ATL0 ---> Disable line feed
> ATS0 ---> Disable sending spaces between hex characters
> ATH1 ---> Turn on odb2 header
> ATCAF1 ---> Turn on CAN auto formatting
> ATSH79B ---> Set request message header 79B
> ATFCSH79B ---> Set flow control header (CAN ID)
> ATFCSD300000 ---> Set flow control data
> ATFCSM1 ---> Set flow control mode; 1 = User defined CAN ID and data
> 
> #######
> ####### Data commands
> #######
> ATSH6A2 ---> Set the header of transmitted OBD messages to header 6A2
> ATFCSH6A2 ---> Set flow control header (CAN ID)
> ATCRA682 ---> Set the CAN hardware filter to 682
> 22D4341 ---> Ambient temperature
> 22D8EF1 ---> Battery temperature
> 22D4021 ---> Speed
> ATSH6B4 ---> Set the header of transmitted OBD messages to header 6B4
> ATFCSH6B4 ---> Set flow control header (CAN ID)
> ATCRA694 ---> Set the CAN hardware filter to 694
> 22D8161 ---> Battery current
> 22D8151 ---> Battery voltage
> 22D86F1 ---> Battery minimum voltage
> 22D8701 ---> Battery maximum voltage
> 22D4101 ---> SOC calibrated
> 22D8651 ---> kWh available
> 22D8601 ---> SOH
> 22D8101 ---> SOC
> ```
> 
> And the already decoded values:
> SoC calibrated (22D4101) = value / 512
> bat min voltage (22D86F1 ) = value in mV
> bat max voltage (22D8701) = value in mV
> 
> So it looks like it's easy to get the data but not all of them are decoded yet.
> 
> I just looked at one of the car files, lets just take the ioniq:
> https://github.com/EVNotify/EVNotify/blob/f87449c7fdf4fe868c48c0bec13a5df62b8f22b5/app/www/components/cars/IONIQ_BEV.vue#L9-L17
> 
> 
> If i take one of those car files as base to get data from one of the PSA cars, what should i enter in that initialization section to get the data from the above mentioned values? If anyone can help me setup a basic car file for PSA cars i can try to decode some more values.

Magneticdud, 2022-01-26:
I tried with a vlink elm327 ble obd adapter with car scanner and I am able to read all the data without any adapter... but the car doesn't give any data when off, even when charging, nullifying the purpose of evnotify. This is even a bigger problem now that PSA pushed some ota update to the telemetry and the car doesn't report to the server at all when charging, so we have no way, official or not, to know the state of charge when charging... this doesn't make sense...

gotgot1995, 2022-02-01:
You can now use any OBD dongle directly with ABRP on most PSA cars. That solved it for me. 😊

hutje, 2022-07-25:
> You can now use any OBD dongle directly with ABRP on most PSA cars. That solved it for me. 😊

Does this mean PSA cars now use standardized OBDII PID? Or is ABRP using the specified PIDs? Because I would like to use OVMS with my PSA car, but currently it is not yet supported. But it supports also standardized OBDII PID. Would like to know, otherwise I need figure out to implement the PID to the special programing structure of OVMS.

ramack, 2025-01-06:
I guess the SoC is also in CAN frame 0x3a8 (bytes 3 and 4 in 1/256%), so byte 3 is the rounded down in full percent. At least this is how it is in my Fiat e-Doblo, which is a relabeled Peugeot e-Rifter. So to read the SoC you don't need those diagnostic services (which will not be answered without "ignition").