EVNotify/EVNotify #207: Cirtoen c-Zero support

HenriApperloo, 2020-09-26:
Citroen c-zero is not supported, don't find the car in the list. Note that this car is the same as Mitsubishi i-Miev and the Peugeot iOn
I got the car, let me know how I can help. I have programming experience (c# .net).

Thanks

GPlay97, 2020-09-29:
If I have the desired PIDs to read out the soc for example, it would be helpful. Otherwise a lot of reverse engineering needs to be done.

HenriApperloo, 2020-09-29:
Hello,

 

I have found the next information about the Triplet Cars (Peugeot Ion, Citroën C-Zero and Mitsubishi i-Miev)

 

http://cantriplet.bitplan.com/

http://can4eve.bitplan.com/index.php/Cantriplet

https://github.com/harite/i-miev-obd2

https://github.com/openvehicles/Open-Vehicle-Monitoring-System/blob/master/vehicle/OVMS.X/vehicle_mitsubishi.c

 

Kind regards,

 

Henri Apperloo

GPlay97, 2020-12-03:
@HenriApperloo Can you write me a mail to info(at)evnotify.de?