EVNotify/EVNotify #208: app stops working in the background

zorkol, 2020-10-04:
### New issue
- Short description: 
The android app stops working after around 20 minutes while in the background.
- Detailed description:
The android app stops working after around 20 minutes while in the background. after manually closing the app and then reopening it it starts working again.

I have it on a pixel 4 with android 11. I have removed the app from android battery management so it is allowed to run all the time. Also tried running in screen on all the time, same issue. 

I have this dongle :Veepeak Mini Bluetooth OBD2 scanner plugged in a kia e-niro 64kw

mhantsch, 2020-10-07:
@zorkol I noticed the same; I think it must have to do with Android's power management. I managed to work around this by running an 'Automate' flow to send EVNotify to the foreground every 30 seconds or so. This seems to keep it busy and prevent Android from trying to hold it.

dimzendk, 2020-11-29:
> @zorkol I noticed the same; I think it must have to do with Android's power management. I managed to work around this by running an 'Automate' flow to send EVNotify to the foreground every 30 seconds or so. This seems to keep it busy and prevent Android from trying to hold it.

What blocks do you use to do that? :)

zorkol, 2020-12-01:
I fixed it by getting in to advanced modes by ticking couple of times in
software number and enabling run in background

mhantsch, 2020-12-01:
> What blocks do you use to do that? :)

I use the 'Start app' block.

My flow does more. See here:
https://e.pcloud.link/publink/show?code=kZ5YkZa9m0IDLmfTSIio2APbo8vk3J3QOX

If you understand German, I have documented my solution here:
https://www.goingelectric.de/wiki/EVNotify-Alternative-Loesung-mit-Automate/

dimzendk, 2020-12-03:
> I fixed it by getting in to advanced modes by ticking couple of times in software number and enabling run in background
> […](#)

Tried that, didn't help on my device :(

GPlay97, 2020-12-03:
You should enable background mode and persistent notifications within the secret settings. Don't forget to save then.

dimzendk, 2020-12-03:
> You should enable background mode and persistent notifications within the secret settings. Don't forget to save then.

Allready done and done :)
But after 20-30min, it stops talking to the dongle.
Fix seems to be to restart the app :/

GPlay97, 2020-12-11:
With all dongles I have, it's running endless with these settings, as long as car is completely on or charging (otherwise it will shutdown, this is wanted behavior to prevent empty 12v batteries). I only have one dongle with a "power off button" - this one goes to deep sleep after 30 minutes and only wakes up again, when I press the button.

dimzendk, 2020-12-11:
I have 2:
https://www.amazon.de/-/en/dp/B071D8SYXN/
https://www.obdsoftware.net/scantools/obdlinkmxplus
Both goes to sleep after a while, when the car is shut off, so they don't drain the 12v battery.
What dongle do you have?

Got a M5stack with EVdash (https://github.com/nickn17/evDash) it reconnects with no problem ,when I turn on the car again with both dongle's.

GPlay97, 2020-12-11:
A few are listed here: Compatibility · https://github.com/EVNotify/EVNotify/wiki/Compatibility

However, most of them are no longer available unfortunately :(

There is "PlugAndPlay" solution: shop.evnotify.de - for EU shipping contact me via mail (outside EU or Germany not possible yet).

Using other software might affect the experience with EVNotify, since often they seems to change more settings, than EVNotify can handle.

mhantsch, 2020-12-11:
@GPlay97 I found that from Android 10 onwards EVNotify stops sending its data after a few minutes even if background mode and persistent notification have been turned on in the extended settings. I scripted around this limitation using an Automate script that forces EVNotify into the foreground every 20 or so seconds (it keeps "starting" the app again and again). With that, it works. Without that, after a few minutes I stopped getting data. This had nothing to do with the dongle.

GPlay97, 2020-12-11:
Good hint @mhantsch . Thank you. 

My devices use Android 9.

spot2000, 2021-01-17:
I have the same problem, do you @GPlay97 have a plan or a hint for this or could you @mhantsch give me a hint on how to do the script so I also can continue with the EvNorify in the background?

mhantsch, 2021-01-17:
> @mhantsch give me a hint 

See the links I posted in an earlier comment here.