EVNotify/EVNotify #210: Need a mean to access % of battery with a python script

Toug19, 2020-12-03:
Hi, this is not a bug report, but a request.
Can I retrieve the % of battery with a python script?

Thanks.

GPlay97, 2020-12-03:
Yes. Currently the API is completely rewritten.

But for now you can access it via GET https://app.evnotify.de/soc?akey=yourAKey&token=yourToken

Toug19, 2020-12-05:
Hi,
Sorry but the link you provided: https://app.evnotify.de/soc?akey=yourAKey&token=yourToken
is not working.

Toug19, 2020-12-05:
Hi,
I tried and it doesn't work:
Here is the command typed with an ubuntu computer (hidden credentials):
`toug@toug-HP-EliteBook-8570p:~$ GET https://app.evnotify.de/soc?akey=xxxxxx&token=xxxxxxxxxxxxxxxxxxxx
[1] 3919
toug@toug-HP-EliteBook-8570p:~$ {"error":{"code":1100,"message":"Missing or invalid parameters."}}
[1]+  Exit 1`
I used copy paste from my web browser (tab settings, credentials) for the akey and token.
The command GET is for Linux or Windows command line?

GPlay97, 2020-12-11:
No. You need to send a HTTP GET Request.. Like your browser does.

You can use `curl` for example. `curl https://app.evnotify.de/soc?akey=yourAKey&token=yourToken`.

Since it's working and the answer of your question was provided, I'll close the issue. If you still have issues, research for HTTP GET.

closed 2020-12-11 by GPlay97

Toug19, 2020-12-11:
Here is the command I type:
curl https://app.evnotify.de/soc?akey=xxxxxx&token=xxxxxxxxxxxxxxxxxxxx
and here is the returned value:
[1] 13629
toug@toug-HP-EliteBook-8570p:~$ {"error":{"code":1100,"message":"Missing or invalid parameters."}}
So I don't see where and how it is working.

I have some knowledge in computer sciences (more low level) but none in HTTP. So maybe there is something.

GPlay97, 2020-12-11:
You need quotation marks. I missed them. Please research the syntax. And use token, not password (just in case this is the issue). I also tried it just now. Working. Getting back JSON response, as it should be.

Toug19, 2020-12-11:
Ok great it works. I have already using the token.
Thanks.