EVNotify/EVNotify #213: 2019 Hyundai Kona 64 kWh cannot connect to EVNotify via OBD dongle

rupoase, 2021-01-07:
### New issue
- Short description: Using a recommended dongle from issue #56 the car cannot connect to the app
- Detailed description:
I have bought [this](https://amzn.to/37MYIQq) dongle which is noted in the compatibility page.
I've also tried the steps mentioned in the issue #56 , as instructed, but still no luck.
I'm using a Huawei P30 lite phone. Torque Lite, Torque Pro works with the car, EVNotify doesn't

- [] critical issue? (_issue, which will affect multiple accounts, 
  an issue which prevents that you can use the app in a normal way,
  bug which makes app unusable, security breach*_) (Yes/No): I don't think so

Errors that can be seen
- [x] There has never been a connection to a car. Please connect your car first time (Invalid date)
- [x] 'Initializing. Please wait' goes into 'Bluetooth failure'

\* **For very critical security breaches, please mail directly at info@evnotify.de**

GPlay97, 2021-01-08:
Bluetooth failed means connection can't be established at all to the Dongle.

Is it a Bluetooth 4.0 one? Note that BLE Dongles only work on iOS.

Besides that - it needs to be paired in the system - then selected within EVNotify settings under OBD2 device and saved.

Note that parallel usage of multiple apps can have a side effect. Restarting the phone without opening the other ones can help then.

rupoase, 2021-01-12:
Hello @GPlay97 ,

The dongle is BT 4.0 as per description (I don't know how to check that one though). 
My Huawei P30 Lite is definitely BT 4.0+ (actually 4.2).
The OBD dongle was firstly paired with my phone, then I've installed the app and configured it.

Only my Kona uses the phone BT. I don't have any other apps that are using the OBD dongle at the same time
I'll try to restart the phone and disconnect the car from the phone, and come back with feedback.

rupoase, 2021-01-12:
I came back from the car, after restarting the phone, the dongle, but it still doesn't connect.
Instead of the red 'bluetooth failed' message, I now receive some message in orange, but the message stays like a split second on the screen, you can't event get to read the first letter. 

Using another app, I can some data that maybe would help you:
- OBD2 protocol: ISO 15765-4 CAN (11 bit ID, 500 kbaud)
- ECU ID: 7EB
- OBD 2 MAC is 24:42:16:08:00:00
- below is the state of the app after the orange message pops up (I've broke the line so that it doesn't take up the whole screen)
![Screenshot_20210112_124939_com evnotify app]
(https://user-images.githubusercontent.com/50882328/104309142-5a542600-54da-11eb-8927-62f81d577e98.jpg)

Maybe it's good to let the message stay at least some seconds on the screen before triggering its dismissal.

Some things that I noticed while using the app, that might need improvements:
- the setting 'Autom. Error tracking' is not being saved after saving, despite receiving the 'Settings saved successfully' message
- I don't get to see or browse the logs of the app. When debugging it says that 'data will be transfered', however I think that the user should at least see the data that's being sent.
- it might be great if I could understand what the state icons on the top-right main page's header mean.
- debug mode icon is sometimes misleading as it doesn't change all the time.

GPlay97, 2021-01-16:
The message means, that the OBD readout process retrieved unexpected data. This means, that the dongle is unfortunately not supported. I'm working on adding a broader support for more dongles soon. The reason for fast disappearing is that it automatically tries to re-initialize. If car is completely on and still doesn't work, the dongle is not initialized correctly unfortunately.

The icons will improve in future versions. The debug mode is not intended for longer usage and was actually used to internally retrieved parsed / interpreted data to improve car models and to expand support for dongles.

Let's see what future updates brings, maybe they help.

rupoase, 2021-02-04:
So, an update about this issue. Definetly this was an OBD compatibility issue.

I've purchased another OBD, the Konnwei KW902 and the application works as expected. However, it seems that it didn't kept open the connection with the OBD device while the application was opened. I wanted to test the integration with ABRP for tracking purposes.
Maybe there's a setting that I didn't noticed.