EVNotify/EVNotify #215: usage on iOS with BT OBD2 dongle

RalphS69, 2021-01-17:
### New issue
- OBD2 dongle and BT connection between app on iOS fails
- the OBD2 dongle does not show up in the App

 <details>
  <summary>
* iPhone 6s with iOS 14.3
* OBD2 dongle: vLinkger MC+ for iOS and Android
* Ioniq elctric classic
</summary>

The starter guide of the OBD2 dongle claims:
"IMPORTANT NOTE
Can not see Bluetooth device name in iPhone?! Don't worry. The device only requre to configure Bluetooth device name in the OBD app setting. Just enable Bluetooth button on the iOS device and continue to the OBD app setting,
ONLY iOS supported Bluetooth profiles can show device name. Bluetooth LE 4.0 technology isn't apply for the suported Bluetooth profiles from iOS, so user can not find vLinkerMC+ Bluetooth device name.
For iOS supported Bluetooth profiles, please refer check this page: https://support.apple.com/en-us/HT204387"

Accordingly, the device is not listed in the iOS list of BT devices. Afaik, the dongle uses Bluetooth LE (4.0).

However, the device shows up in the app "Car Scanner" as "vLinker MC-IOS" and this App shows subsequently the car parameters as expected. This proves, that the dongle works and a BT connection between the dongle and the iPone can be established.

However, the device does not show up in the app "EVNotify", so I cannot get aces to the car parameters using this app.
</details>

burnsfisher, 2021-03-13:
I got mine to work thanks to reading this issue.   I did find the vGate device listed on EVNotify, but it did not seem to pair up correctly.  It would not "stay connected".  I downloaded the above ELM OBD2 scanner, and it paired easily.  Then I went back to EVNotify and it seemed to connect ok.  This was on a US 2019 Chevy Bolt, so there was not a lot of data--mainly the state of charge.  But that's all I wanted.

meitran, 2021-04-17:
I actually got the same issue with iphone 11 pro, ios 14.4.2 and the Veepeak OBDCheck BLE+ Bluetooth 4.0 OBD2 Scanner. This is not a cheap adapter, and the adapter works fine with the app «Car Scanner». In Evnotify the app will not even show on the list.

closed 2021-05-18 by RalphS69