EVNotify/EVNotify #216: OBD2 Bluetooth connection problem

Grasmaaiers, 2021-01-21:
### New issue

I'm trying to test EVNotify with my ID.3, using a standard OBD2-dongle. 
But: my phone (Oneplus 7pro with Android 11) refuses to make a bluetooth-connection. Also a different phone (also running Android11) refuses the connection with the dongle. 

Do you have any advice on what dongle to buy?

I'm driving an VW id.3, so very eager to start testing

GPlay97, 2021-01-21:
We are already in contact - but just here for information. Is Bluetooth connection completely not possible, even though the correct obd2 device is selected within settings from EVNotify (and saved)? So within dashboard you get Bluetooth failed? Or do you just get no valid data (so dongle is connected but not properly handled)?

Bluetooth connection issue could be related to Android 11 - might need to take a look at this asap

Grasmaaiers, 2021-01-21:
ok, more details here. Tried it on 3 Android phones:

Phone 1: (oneplus 7 pro, Android 11): refuses to pair with OBD2
Phone 2: (Google Pixle5 with Android 11): pairs with OBD2. But: EVNotify gets no data ("Bluetooth connection problem")
Phone 3: (Samsung S20, Android 11), refuses to pair with OBD2

So: problem seems to be in the combination of (cheap) OBD2-bluetooth-adapters and certain phones or Android 11

GPlay97, 2021-01-21:
To clarify: Refuses to pair means in Android native settings?

This is independent from my code. Is engine completely on? This could also be a problem

Grasmaaiers, 2021-01-21:
Seems to be a native Android11 problem. Not pairing. Or: pairing and not transfering data to EVNotify. 
Depending on the phone.

Engine on or off is not helping. Tried that also

uahero, 2021-02-15:
Hi, I have a similar issue - I have purchased elm327 based OBD2 dongle on Amazon https://www.amazon.se/Moonar-ELM327/dp/B00WU97RH8 which actually connects to my Android Phone - Samsung A3-6 with Android 7.0 but transfers nothing into the App. 
I assume this should be the issue with the Dongle. If someone is already using EVNotify with ID3 successfully, please advise the model you use

c64emulator, 2021-02-27:
Some of the wiki links for recommended adapters are dead. I bought the OBD2 Bluetooth 4.0 Adapter with power button from amazon. This adapter did not work for me (I have Android 10). So I tested my old huge OBD2 adapter. This adapter is rather old (ten years at least). BUT: this adapter works!

c64emulator, 2021-02-28:
- **working Dongle:**

[image]
[image]

```
> Scan datum=2021-02-28 11:35:28
> Geräte name=OBDII
> Geräte mac=00:06:71:00:00:06
> Die Geräte version (deklariert)=ELM327 v1.5
> 
> Befehl	Version	Ergebnis	RAW
> AT@1	1.0	OK	AT@1\rOBDII to RS232 Interpreter\r\r>	
> ATSPA6	1.0	OK	ATSPA6\rOK\r\r>	
> ATTPA6	1.0	OK	ATTPA6\rOK\r\r>	
> ATTP6	1.0	OK	ATTP6\rOK\r\r>	
> ATSP00	1.3	OK	ATSP00\rOK\r\r>	
> ATSP6	1.0	OK	ATSP6\rOK\r\r>	
> ATAL	1.0	OK	ATAL\rOK\r\r>	
> ATNL	1.0	OK	ATNL\rOK\r\r>	
> ATAMC	2.0	KO	ATAMC\r?\r\r>	
> ATAMT20	2.0	KO	ATAMT20\r?\r\r>	
> ATRA70	1.3	OK	ATRA70\rOK\r\r>	
> ATSR70	1.2	OK	ATSR70\rOK\r\r>	
> ATAR	1.2	OK	ATAR\rOK\r\r>	
> ATAT0	1.2	OK	ATAT0\rOK\r\r>	
> ATAT2	1.2	OK	ATAT2\rOK\r\r>	
> ATAT1	1.2	OK	ATAT1\rOK\r\r>	
> ATBD	1.0	OK	ATBD\r0C 17 FE 00 7B 07 62 18 9D 05 80 07 80 \r\r>	
> ATBRT0F	1.2	OK	ATBRT0F\rOK\r\r>	
> ATCAF0	1.0	OK	ATCAF0\rOK\r\r>	
> ATCAF1	1.0	OK	ATCAF1\rOK\r\r>	
> ATCEA04	1.4	OK	ATCEA04\rOK\r\r>	
> ATCERF1	2.2	KO	ATCERF1\r?\r\r>	
> ATCEA	1.4	OK	ATCEA\rOK\r\r>	
> ATCF00000111	1.0	OK	ATCF00000111\rOK\r\r>	
> ATCF111	1.0	OK	ATCF111\rOK\r\r>	
> ATCFC0	1.0	OK	ATCFC0\rOK\r\r>	
> ATCFC1	1.0	OK	ATCFC1\rOK\r\r>	
> ATCM00000111	1.0	OK	ATCM00000111\rOK\r\r>	
> ATCM111	1.0	OK	ATCM111\rOK\r\r>	
> ATCP18	1.0	OK	ATCP18\rOK\r\r>	
> ATCRA7C0	1.3	OK	ATCRA7C0\rOK\r\r>	
> ATCRA000007C0	1.3	OK	ATCRA000007C0\rOK\r\r>	
> ATCRA	1.4b	KO	ATCRA\r?\r\r>	
> ATCS	1.0	OK	ATCS\rT:00 R:00 \r\r>	
> ATCSM0	1.4b	KO	ATCSM0\r?\r\r>	
> ATCSM1	1.4b	KO	ATCSM1\r?\r\r>	
> ATCTM5	2.1	KO	ATCTM5\r?\r\r>	
> ATCTM1	2.1	KO	ATCTM1\r?\r\r>	
> ATCV0000	1.4	OK	ATCV0000\rOK\r\r>	
> ATD1	1.3	OK	ATD1\rOK\r\r>	
> ATD0	1.3	OK	ATD0\rOK\r\r>	
> ATDP	1.0	OK	ATDP\rISO 15765-4 (CAN 11/500)\r\r>	
> ATDPN	1.0	OK	ATDPN\r6\r\r>	
> ATE0	1.0	OK	ATE0\rOK\r\r>	
> ATE1	1.0	OK	OK\r\r>	
> ATFCSD0430FF00	1.1	OK	ATFCSD0430FF00\rOK\r\r>	
> ATFCSH000007B0	1.1	OK	ATFCSH000007B0\rOK\r\r>	
> ATFCSH7B0	1.1	OK	ATFCSH7B0\rOK\r\r>	
> ATFCSM0	1.1	OK	ATFCSM0\rOK\r\r>	
> ATFE	1.3a	OK	ATFE\rOK\r\r>	
> ATH0	1.0	OK	ATH0\rOK\r\r>	
> ATH1	1.0	OK	ATH1\rOK\r\r>	
> ATI	1.0	OK	ATI\rELM327 v1.5\r\r>	
> ATIB10	1.0	OK	ATIB10\rOK\r\r>	
> ATIB12	2.2	KO	ATIB12\r?\r\r>	
> ATIB15	2.2	KO	ATIB15\r?\r\r>	
> ATIB48	1.4	OK	ATIB48\rOK\r\r>	
> ATIB96	1.0	OK	ATIB96\rOK\r\r>	
[... 139 lines cut ...]
> ATM0	1.0	OK	ATM0\r\nOK\r\n\r\n>	
> ATPPFFON	1.1	OK	ATPPFFON\r\nOK\r\n\r\n>	
> ATPPFFOFF	1.1	OK	ATPPFFOFF\r\nOK\r\n\r\n>	
> ATPPS	1.1	KO	ATPPS\r\n00:FF F  01:FF F  02:FF F  03:32 F\r\n04:01 F  05:FF F  06:F1 F  07:09 F\r\n08:FF F  09:00 F  0A:0A F  0B:FF F\r\n0C:68 F  0D:0D F  0E:FF F  0F:FF F\r\n10:0D F  11:00 F  12:FF F  13:32 F\r\n14:FF F  15:FF F  16:FF F  17:92 F\r\n18:00 F  19:FF F  1A:FF F  1B:FF F\r\n1C:FF F:TIMEOUT?>	
> ATR0	1.0	OK	ATR0\r\nOK\r\n\r\n>	
> ATR1	1.0	OK	ATR1\r\nOK\r\n\r\n>	
> ATRV	1.0	OK	ATRV\r\n15.0V\r\n\r\n>	
> ATS0	1.3	OK	ATS0\r\nOK\r\n\r\n>	
> ATS1	1.3	OK	ATS1\r\nOK\r\n\r\n>	
> ATSDFF	1.4	OK	ATSDFF\r\nOK\r\n\r\n>	
> ATRD	1.4	OK	ATRD\r\nFF \r\n\r\n>	
> ATSH000007E0	2.0	OK	ATSH000007E0\r\nOK\r\n\r\n>	
> ATSHA848F1	1.0	OK	ATSHA848F1\r\nOK\r\n\r\n>	
> ATSH7B0	1.0	OK	ATSH7B0\r\nOK\r\n\r\n>	
> ATST32	1.0	OK	ATST32\r\nOK\r\n\r\n>	
> ATSWFF	1.0	OK	ATSWFF\r\nOK\r\n\r\n>	
> ATTAF9	1.4	OK	ATTAF9\r\nOK\r\n\r\n>	
> ATV1	1.3	OK	ATV1\r\nOK\r\n\r\n>	
> ATV0	1.3	OK	ATV0\r\nOK\r\n\r\n>	
> ATWMC133F13E	1.0	OK	ATWMC133F13E\r\nOK\r\n\r\n>	
> ATPC	1.0	OK	ATPC\r\nOK\r\n\r\n>	
> ATWS	1.0	OK	ATWS\r\n\r\rELM327 v1.5\r\n\r\n\r\n>	
> ATD	1.0	OK	ATD\r\n�OK\r\r>	
> ATZ	1.0	OK	ATZ\r�\r\rELM327 v1.5\r\r>	
```

SlOrbA, 2021-04-11:
I've also unable get data with the same power button dongle. There has been pairing (ID.3 Pro S) and initialization with evnotify, but no data readings.

c64emulator, 2021-04-12:
The old (working) OBD-Dongle has some drawbacks:

- false alarms from alarmsystem. 
- lack of security (pairing is easy, password is well known)
- the performance, some messages get lost (buffer overflow)
- firmware not able for updates in the field
- size

Therefore in the meantime I use the "OBDlink LX": smaller, more MIPS (performance),  more flash memory (for software updates), more RAM (for messages buffers), field updateable, security by pairing button. Since then I had no false alarms.
Compare yourself (ELM327 has Microchip-PIC18F2480 (8-bit, 4MIPS, 32KB ROM, 1.5KB RAM,) "OBDlink LX" has STN1110 / Microchip-PIC24HJ128GP502 (16-bit, 40MIPS, 128KB ROM, 8KB RAM), "OBDlink MX+" has STN2120 / Microchip-dsPIC33EP256GP504 (16-bit, 70MIPS, 256KB ROM, 32KB RAM)):  
[image]

I bought it together with a OBD2-extension-cable, so removing or changing dongles or observing the signal LEDs is easier. 
[image]

SlOrbA, 2021-04-14:
> I've also unable get data with the same power button dongle. There has been pairing (ID.3 Pro S) and initialization with evnotify, but no data readings.

I could be that there is some connection problem with the OBD2 port. I've now unplugged and plugged the dongle many times and now it has started to get data and work steadily.

marco-gallinari, 2021-06-24:
>  "OBDlink LX" 

Is it working good?

I've just returned another kind because It wasn't compatibile with my ID3, and came here to look for the best option... I was dolly enough thinking any BT dongle would do the work :)

kristgy, 2021-06-25:
Hi Marco,
I am using this cheap noname dongle with my Hyundai Ionic for a while now
with good results with EV notify on Android 11:
https://www.amazon.de/gp/product/B07PLDC2SC/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1
I like the fact that it has a physical power button so it is easy to switch
it off completely when leaving the car (making your OBD2 port accessible
wirelessly like this is  probably not great from a security perspective).
The challenge with buying dongles online is that most of them contain
copycat chips with unknown compatibility.
Best,
kristgy

marco-gallinari, 2021-06-25:
Hi kristgy, 

thank you for sharing.

Strange enough, I just returned this item:

https://www.amazon.de/gp/product/B07FMFRLN3/

because it does not work; unfortunately, I didn't read in the last lines of the description that EVs are not supported.

Are you sure this one works? They look basically the same :)

kristgy, 2021-06-25:
It has been running solid for a couple of months now. I use it every day in
combination with ABRP.
My setup is Hyundai Ioniq 28 kWh and Google Pixel 4a running Android 11.
The initial setup was a bit fiddly and there is very little useful
documentation.
The first step is to pair the dongle with the phone (the PIN is 1234 or
12345, if I remember correctly) and then to set the correct BT interface in
EV notify.
The challenge for end-user software like EV Notify is that there are so
many different ways this can fail, given the many possible combinations of
dongle, phone and car.

For example: Are you on iOS or Android?

- kristgy

marco-gallinari, 2021-06-25:
I'm on Android, on a OnePlus Nord.
I am trying to connect to a vw Id3.
The dongle connects to the phone, EVNotify recognise It, but no data is read.
I tried with other apps like car scanner, same thing: connects, but data Is empty.
Then I read the final note on Amazon description :)

Grasmaaiers, 2021-06-25:
Guys, stop the search. It depends what phone you're using. I also had no success using a OnePlus phone, but other android works fine

marco-gallinari, 2021-07-03:
I purchased a Vgate iCar Pro BT dongle, and now everything works properly.

Only thing is, when I leave the car, I have to pull the dongle out and then back in to reconnect, maybe due to some issue with BLE power saving I suppose

acrollet, 2021-07-03:
> I purchased a Vgate iCar Pro BT dongle, and now everything works properly.
> 
> Only thing is, when I leave the car, I have to pull the dongle out and then back in to reconnect, maybe due to some issue with BLE power saving I suppose

This is the problem I have - has anyone figured out how to get android to reconnect without physically removing and re-inserting the dongle?

GPlay97, 2021-07-04:
> This is the problem I have - has anyone figured out how to get android to reconnect without physically removing and re-inserting the dongle?

VGate iCar Dongle is not intended to be used again without repluging. I use this one for Android currently: https://amzn.to/3fjjyxC

c64emulator, 2021-07-04:
> I purchased a Vgate iCar Pro BT dongle, and now everything works properly.
> 
> Only thing is, when I leave the car, I have to pull the dongle out and then back in to reconnect, maybe due to some issue with BLE power saving I suppose

see #187